uraimo/SwiftyGPIO #3: Does it work on a BeagleBone Black?

uraimo, 2016-01-12:
Does it work on a BeagleBone Black?

This task mainly involves general testing and verification of the GPIO ids.

uraimo, 2016-01-12:
For those than can do a quick test https://gist.github.com/uraimo/b879bfda00470447fe2f

The [beaglebone pinout](https://github.com/uraimo/SwiftyGPIO/wiki/GPIO-Pinout) is not yet added to the library defaults, so, these lines:

    let gpios = SwiftyGPIO.getGPIOsForBoard(.RaspberryPiB2Zero)
    var gp = gpios[.P2]!

should be replaced with:

    var gp = GPIO("GPIO_20",20)  // This should work, it's pin 41 in the P9 header

hpux735, 2016-01-12:
I got a weird response on the Beaglebone Black:

```wdillon@arm:~$ cat main.swift 
import Glibc

let gpios = SwiftyGPIO.getGPIOsForBoard(.RaspberryPiB2Zero)
var gp = GPIO(name: "GPIO_20",id: 20)  
print("Current Status")
print("Direction: "+gp.direction.rawValue)
print("Edge: "+gp.edge.rawValue)
print("Active Low: "+String(gp.activeLow))
print("Value: "+String(gp.value))
gp.direction = .OUT
gp.value = 0 // Or 1 if the previous value is 0
print("New Status")
print("Direction: "+gp.direction.rawValue)
print("Edge: "+gp.edge.rawValue)
print("Active Low: "+String(gp.activeLow))
print("Value: "+String(gp.value))
wdillon@arm:~$ swiftc SwiftyGPIO/Sources/SwiftyGPIO.swift main.swift 
wdillon@arm:~$ ./main 
Current Status
fatal error: unexpectedly found nil while unwrapping an Optional value: file /home/wdillon/swift/stdlib/public/core/Optional.swift, line 122
Illegal instruction
wdillon@arm:~$ 
```

uraimo, 2016-01-12:
Looks like it didn't create the gpio in /sys/class/gpio/ , should fail more gracefully.

Looking at the documentation it looks like that the GPIO Ids have a base number that needs to be added to obtain the real sysfs id, i'm looking into this.

hpux735, 2016-01-12:
Yah, the GPIOs on Beaglebone are weird relative to raspis.

uraimo, 2016-01-12:
Ok, finding the gpio id of a pin is not that easy with the Beagle Bone :smile: 

With 02caacace42696f519773d0ee58da44f68f5e008 i've committed an initial set of gpios, built selecting only the pins that are gpios from boot without being configured (i guess, it looks like others needs to be configured enabling a specific mode, descriptions of the pins taken [from here](https://github.com/CircuitCo/BeagleBone-Black/blob/master/BBB_SRM.pdf?raw=true)).

Regarding your test, the GPIO with id 20, is [shown in white here](https://github.com/uraimo/SwiftyGPIO/wiki/GPIO-Pinout) , i don't know what it means but let's just say it was a bad choice and i should've chosen another pin.

The innocuous GPIO45 (PIN11) could be a better choice.

hpux735, 2016-01-12:
Beaglebone works.  See this PR: https://github.com/uraimo/SwiftyGPIO/pull/6

It still needs a pin map, and to be honest GPIO on the BBB is kinda a nightmare.

uraimo, 2016-01-13:
I added an initial set of pins taken from the BeagleBone Reference, but i'm not too sure if it's  the best approach:

```
    static let GPIOBEAGLEBONE:[GPIOName:GPIO] = [
        .P0:GPIO(name:"P8_PIN03_GPIO1_6",id:38),  //P8
        .P1:GPIO(name:"P8_PIN04_GPIO1_7",id:39),
        .P2:GPIO(name:"P8_PIN05_GPIO1_2",id:34),
        .P3:GPIO(name:"P8_PIN06_GPIO1_3",id:35),
        .P4:GPIO(name:"P8_PIN11_GPIO1_13",id:66),
        .P5:GPIO(name:"P8_PIN12_GPIO1_12",id:67),
        .P6:GPIO(name:"P8_PIN14_GPIO0_26",id:26),
        .P7:GPIO(name:"P8_PIN15_GPIO1_15",id:47),
        .P8:GPIO(name:"P8_PIN16_GPIO1_14",id:46),
        .P9:GPIO(name:"P8_PIN17_GPIO0_27",id:27),
        .P10:GPIO(name:"P8_PIN18_GPIO2_1",id:65),
        .P11:GPIO(name:"P8_PIN20_GPIO1_31",id:63),
        .P12:GPIO(name:"P8_PIN21_GPIO1_30",id:62),
        .P13:GPIO(name:"P8_PIN22_GPIO1_5",id:37),
        .P14:GPIO(name:"P8_PIN23_GPIO1_4",id:36),
        .P15:GPIO(name:"P8_PIN24_GPIO1_1",id:33),
        .P16:GPIO(name:"P8_PIN25_GPIO1_0",id:32),
        .P17:GPIO(name:"P8_PIN26_GPIO1_29",id:61),
        .P18:GPIO(name:"P8_PIN27_GPIO2_22",id:86),
        .P19:GPIO(name:"P8_PIN28_GPIO2_24",id:88),
        .P20:GPIO(name:"P8_PIN29_GPIO2_23",id:87),
        .P21:GPIO(name:"P8_PIN30_GPIO2_25",id:89),
        .P22:GPIO(name:"P8_PIN39_GPIO2_12",id:76),
        .P23:GPIO(name:"P8_PIN40_GPIO2_13",id:77),
        .P24:GPIO(name:"P8_PIN41_GPIO2_10",id:74),
        .P25:GPIO(name:"P8_PIN42_GPIO2_11",id:75),
        .P26:GPIO(name:"P8_PIN43_GPIO2_8",id:72),
        .P27:GPIO(name:"P8_PIN44_GPIO2_9",id:73),
        .P28:GPIO(name:"P8_PIN45_GPIO2_6",id:70),
        .P29:GPIO(name:"P8_PIN46_GPIO2_7",id:71),
        .P30:GPIO(name:"P9_PIN12_GPIO1_28",id:60),   //P9
        .P31:GPIO(name:"P9_PIN15_GPIO1_16",id:48),
        .P32:GPIO(name:"P9_PIN23_GPIO1_17",id:49),   //Ignore Pin25, requires oscillator disabled
        .P33:GPIO(name:"P9_PIN27_GPIO3_19",id:125)
    ]
```

I'll have to read the reference more thoroughly to understand if those are really the pins configured as gpios from boot (without the need for additional configuration). I suppose the BBB can change the mode of some of the pins setting some registers with specific bitmasks (like you do with traditional embedded boards) but at the moment the library is too simple to support something like that.

uraimo, 2019-06-25:
Obsolete issue, closed.

closed 2019-06-25 by uraimo