uraimo/SwiftyGPIO #5: Error return value not checked in export

hpux735, 2016-01-12:
From strace:

open("/sys/class/gpio//export", O_WRONLY|O_CREAT|O_TRUNC, 0666) = 3
futex(0x76f62828, FUTEX_WAKE_PRIVATE, 2147483647) = 0
fstat64(3, {st_mode=S_IFREG|0770, st_size=4096, ...}) = 0
mmap2(NULL, 4096, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x74d8a000
write(3, "2", 1)                        = -1 EBUSY (Device or resource busy)
close(3)                                = 0
munmap(0x74d8a000, 4096)                = 0

hpux735, 2016-01-12:
But, I think this just happens when it's already exported.  So, I'm not sure whether it matters.

uraimo, 2016-01-12:
Thanks for opening the issue, i could test it only on the CHIP (Debian jessie) and there the double export was not an issue, but that board is a bit different from the Pi.

Did it work the first time you tried it? (or alternatively, can you do a *echo "2" > /sys/class/gpio/unexport* and try again? replacing that 2 with the GPIO is you used)

hpux735, 2016-01-12:
The funny thing is that I forgot that the Raspi v.2 is at home (and therefore not connected to an oscilloscope) the Raspi v.1 is here at work.  I bet it was working, but of course I didn't see anything! lol.  Anyway, I'm getting ready to test on the v.1.

uraimo, 2016-01-12:
Even better! I tried do to a quick test on an old Pi A Rev 1 using iachievedit binaries but swiftc segfaulted (makes sense, it's not an arm7), i'll try with your first tgz.

hpux735, 2016-01-12:
My tar.gz is armv7, too.  I'm building a new armv6 one.  Standby.

uraimo, 2016-01-12:
I confirm that what you saw happens when a gpio is already exported, but it doesn't have any side-effects. 

Anyway, to help debugging future issues, when fwrite/fread reports an error (does not happen in that case) a message will now be displayed and then the program will abort (added with c038e7d8e7fd757a2f4e5f77bb1d2c394fae68cc).

closed 2016-01-15 by uraimo