uraimo/SwiftyGPIO #8: Adding support for hardware PWM

uraimo, 2016-01-15:
PWM outputs have a few uses (for hobby projects mainly driving motors and playing sounds) and a convenient sysfs interface. Implementation should be really straightforward.

ClaudioTSM, 2016-01-21:
Hello @uraimo. 
I'm planning a robotic product with swift/rpi stack. To drive motors and movements having PWM is essential. I was wondering how much priority have you given to supporting it.
Thanks for the great work.

uraimo, 2016-01-21:
Hi Claudio,
in the last few days i've been building a library to drive the PCD8544 lcd and considering how slow the Swift software SPI implementation turned out to be the item with higher priority is the sysfs SPI interface (likely not much work).
After that, considering that i don't have enough info yet to implement the transition from sysfs to mmap of the current functionalities i could start implementing some form of PWM.
The implementation doesn't seem that hard, verification and testing before release could need more time instead.

There are a few alternatives regarding what kind of PWM to expose from Swift:

1. One (or two) hardware PWMs: Are accessible via sysfs, board-independent approach, fastest PWM that the Raspi&co can generate.

2. Libraries like Servoblaster/PiBlaster provide 8 DMA driven PWMs: Still accessible via sysfs/dev with a custom interface, the libraries support the Raspi & BBB but not other boards AFIAK, not as fast but looks like they are more than fast enough for servos.

3. Support for external PWM driver boards via I2C/SPI[1]: Harder to implement, harder to test but give access to more, fast, PWM outputs. But doesn't look like something that should be integrated in SwiftyGPIO.

So considering that SwiftyGPIO doesn't target the Raspi specifically, i'll start with (1), that will work for more than one kind of board, and then i'll consider introducing (2) as an extension (we will depend on those libraries, at the moment i don't see any reason to reimplement the same stuff in Swift).

How much time to have something usable? Possibly, sometime next month.

[1] SPI is supported, for an I2C Swift library see https://github.com/Sephiroth87/SMBus-swift

ccarnino, 2016-01-21:
This is very interesting! 
I will be happy to try the do stuff with the integrated hardware PWMs. I will later try to use the driver boards with the SMBus-swift library.
I will keep an eye on this. Maybe I will be able to test it.
Cheers

uraimo, 2016-06-14:
Obligatory ~6 months update, i didn't forget but other things took precedence, i tested servoblaster a few months ago and it works, but i'm still thinking about implementing something similar in Swift.

Another thing that will come with this, and these features will make up most of v1.0, is pattern based output leveraging the servoblaster-like dma based pwm.

ccarnino, 2016-06-14:
Thanks for the update!

hariseldon78, 2016-12-11:
Hi, any news on this feature?

pgorzelany, 2017-02-28:
Bump. Any news on this? Also, is there a way using SwiftyGPIO to simulate PWM with software?

uraimo, 2017-03-01:
Good News Everyone, I've finally an initial PoC that works.

This first version will only expose the 2 hardware PWM outputs that are available on all Pis. I don't exclude that when I'll be done with this I could be able to implement the 8 DMA-based PWMs as ServoBlaster did.

I plan to test this with one of those SG90 servos and with all the WS2812 stuff I've bought and never used.

More details soon.

uraimo, 2017-03-08:
Ok, I've just pushed the release 0.8.9 that introduces the PWM.

I've tested it with signals with a period from 2-300ns to 200ms and the results are acceptable. Even if both PWMs work, we still can't use them at the same time, there is a minor issue that needs to be fixed.

If anyone could do further testing on this I'll appreciate it (device or oscilloscope), I haven't tested it yet with the Raspberry 1 or with systems that have `/dev/gpiomem`.

At the moment I'm building a simple library for the SG90.

uraimo, 2017-03-08:
Ok, done: https://github.com/uraimo/SG90Servo.swift

uraimo, 2017-03-22:
Ok, I've added what was needed to drive WS2812 leds (led strips, led matrixes, etc...), a new PWM mode that generates signals using bit-pattern sequences. ATM there is just an example with a few boring effects, documentation and a comfortable WS281x driver to follow soon.

It's still not possible (for classic PWM and bit-pattern PWM generator) to really use two PWM outputs together.

uraimo, 2017-03-23:
Simple effect with 124 leds, one 60 led strip connected serially to a 64 led matrix.

[image]

uraimo, 2017-03-24:
[image]

[Higher resolution](https://giant.gfycat.com/UltimateAgileBeardeddragon.mp4)

closed 2017-03-24 by uraimo