uraimo/SwiftyGPIO #10: Testing on Raspberry Pi A, B, A+, B+, Zero (ARM6)

uraimo, 2016-01-17:
Since some tests have been done on RPi 2, i'm moving that among the tested boards.

This issue is related to the ARM6 Raspberry boards, AFAIK there are no ready to install packages for those yet (also, those boards don't run ubuntu and Raspbian doesn't have clang-3.6, required to build Swift, so that needs to be built too). I'm trying to build everything on an old Rev1, i know others are already working on it with more chances to see the end of it.

A simple GPIO read/write test on one of these board could be enough to understand if there are major issues, other than that it could be nice, but not essential, to verify if all the defaults GPIOs are correct.
To sum up the GPIO presets for those boards:  

* SupportedBoard.RaspberryPiRev1 (A and B Revision 1 boards):  The boards built before September 2012, with a 26 pins header (too old?)
* SupportedBoard.RaspberryPiRev2 (A and B Revision 2 boards): The boards built after September 2012, still with 26 pins header but with slightly different gpio ids
* SupportedBoard.RaspberryPiPlus2Zero (A+, B+, the new Zero): The boards with the 40 pins header.

hpux735, 2016-01-17:
Bummer, it looks like I shutdown my v.1.  While swift should be built with Clang-3.6, I was able to build it with the older version that is in raspian jessie light.  It just takes several days, and you'll need lots of swap (no news there).

Also, I don't have a Rev1A.  I do have a Rev1B, and B+.

When I get back to work on Tues, I'll try to remember to restart but build for the ARMv6, and I'll upload an image to my blog.

uraimo, 2016-01-17:
> I was able to build it with the older version that is in raspian jessie light.

Really a GOOD news (i read somewhere that they used some c++14 features that needed 3.6, i'm glad it's not true), i'm compiling 3.6 right now on a ARev1 (...) and i expect it to fail in a 10 hours or so. I'll try compiling directly Swift with 3.5 then. If all this fails in a few days(likely) i'll gladly use your images :)
To speed it up i've moved the source/builddir and the swap to an usb drive, huge improvement.

hpux735, 2016-01-17:
Yep, that's a good move (the usb drive).

I suspect that the Clang-3.6 thing might be a cause of some weirdness on ARMv6, but, I remember it at least building and running the hello world and some Foundation stuff.

uraimo, 2016-02-12:
Closing this, it works.

closed 2016-02-12 by uraimo