uraimo/SwiftyGPIO #12: Testing on Tegra TK1

uraimo, 2016-01-22:
It needs to be tested and it would be nice to have a list of the GPIO ids.

dskat, 2017-04-13:
Are you still wanting help on testing Tegra TK1 or even TK2?

uraimo, 2017-04-13:
YES, that would be great.

dskat, 2017-04-13:
Okay great. I will need to get myself oriented with the project and have this installed on one of our Tegra boards.

dskat, 2017-04-13:
Do you have any formal guidelines for testing?

uraimo, 2017-04-13:
Thanks a lot Eli, let me know if you need any info or help.

If you are using Debian this are likely the Swift binaries that are more likely to work: http://www.housedillon.com/other/swift-armv7-3.0.2.tar.gz

uraimo, 2017-04-13:
Nothing formal i'm afraid (but that's a good idea, it will simplify a lot of things), I usually recommend to test a basic sysfs gpio example like this one: https://github.com/uraimo/SwiftyGPIO/blob/master/Examples/CustomGPIO/Sources/main.swift

but using the correct gpio ids for the board under test

dskat, 2017-04-13:
Thanks for the info. I will keep you posted in the coming weeks on my progress.

uraimo, 2019-06-25:
Obsolete issue, closed.

closed 2019-06-25 by uraimo