uraimo/SwiftyGPIO #16: Libraries based on SwiftyGPIO, some ideas

uraimo, 2016-02-24:
This issue collects interesting libraries that could be built from SwiftyGPIO and that could be useful for other people building embedded projects with Swift using specific sensors/devices, feel free to chime in if you want to build one of these or if you have other (better) ideas/requests:

- MAX7219, for both led matrix and seven segment arrays based on this driver (uses SPI).
- HMC8553L Magnetometer, requires I2C(N/A).
- SSD1306 / SH1106 128x64 OLED, sometimes available with SPI other times with I2C(N/A).
- ST7735 TFT color LCD, uses SPI.
- Wii Nunchuk, uses I2C(N/A).
- WS281X RGB LED strips, requires PWM(N/A) or can be done manually with bit banging.
- APA102/C RGB LED strips, uses SPI.
- nRF24L01 radio module, uses SPI.
- DCF77 radio time receiver, custom protocol.
- DS130X real time clock, custom protocol.

damuellen, 2016-02-29:
I have bought an Raspberry Pi2 and Unicorn HAT (8x8 WS2812 LEDs) some weeks ago.
Which I would like to see blink, i played around with a C Lib and Swift but it didn't worked out the way I want. Only 5 LEDs light up. So I would prefer a Swift only solution with SwiftyGPIO. But I need some help, this low level stuff is unchartered terrain for me.

richardxyx, 2016-03-01:
I see potential in a WS2812 library!
My horrible solution consists of calling the Python library from Swift right now.

uraimo, 2016-03-01:
@damuellen i'm not sure it could be related, but are you converting the raspberry 3v3 gpios to 5v?
 
I can't find much documentation about this but i suppose the Unicorn Hat should be like a strip of 64 leds in series, so i *suppose* we could make it work sending a sequence of 1536 impulses (64 * 8bit * rgb) manually from a gpio respecting the WS2812 timing:

[image]

This could be easily done setting *gpio.value* and then nanosleep*ing* for the time described above, al this repeated in a loop that sends the 1500+ bits and then send the reset signal to apply the configuration (0 for more than 50 micros). For example, to send a single bit at 1:

func sleepnanos(nanos:UInt32){
  var t:timespec = timespec()
  t.tv_sec = 0
  t.tv_nsec = nanos
  nanosleep(nanos,nil)
}

gpio.value = 1
sleepnanos(UInt32(700))
gpio.value = 0
sleepnanos(UInt32(600))

It ***could*** be fast enough and precise enough to do something meaningful (the colors could be a bit or a lot off if the timing is not precise).
I just bought a NeoPixel matrix that should be more or less similar to what you have, to play a bit with it.

@richardxyx yep, i agree :smile: . What kind of ws2812 device do you have?

More info [here](https://learn.sparkfun.com/tutorials/ws2812-breakout-hookup-guide)

richardxyx, 2016-03-01:
@uraimo I'm currently on a project that involves 115 WS2812b 5050 LEDs (Adafruit calls them NeoPixels but they're significantly cheaper from China) from a LED Strip.
I still have some at home for testing and also have some 74AHCT125 to convert the signal from 3V3 to 5V.

It is well known that the WS2812 LEDs require pretty strict timing and I could only find one libary that was actually working with my Raspberry Pi 2 (https://github.com/jgarff/rpi_ws281x).

uraimo, 2016-03-01:
@richardxyx great, so we'll be able to test it on a longer strip too.

I've created a new repo at https://github.com/uraimo/WS2812_RGBLCD.swift with an initial implementation, i'll be able to test it at the end of the week, but i don't expect that this code written blindly should work :) , feel free to play around with it if you want guys.

So, for the timing constant i'm using this for the [WS2812](https://www.adafruit.com/datasheets/WS2812.pdf) and this for the [WS2812B](https://www.adafruit.com/datasheets/WS2812B.pdf). 

Now that i look at the reference more carefully it looks like that the data sent needs to be continuously refreshed after the reset delay, so, to work that `setColor` should be enclosed in a loop... not really an ideal implementation...

richardxyx, 2016-03-01:
I'll try my best helping. I'll play around with the repo, looks like you did a nice job setting everything up.

uraimo, 2017-05-10:
WS281X done: https://github.com/uraimo/WS281x.swift
DS130X I2C done: https://github.com/uraimo/DS1307.swift

damuellen, 2017-05-10:
WS281x.swift works like a charm for me, so thanks a lot for this great lib. Easter holidays was long enough to write some Swift code, that let me control my 🦄Hat with an 📱 app. 😎

uraimo, 2017-05-10:
Thanks @damuellen, glad to hear that someone is using it! :)

jrahaim, 2018-03-26:
I got the AdaFruit 8x8 via HT16K33 working this weekend:

https://github.com/jrahaim/swift-raspberry-pi-adafruit-led

uraimo, 2018-03-26:
Nice one @jrahaim , I've added it to the libraries section.

jrahaim, 2018-03-26:
Also added  and tested the Adafruit 4 digit 14 segment Alpha numeric display to https://github.com/jrahaim/swift-raspberry-pi-adafruit-led.

Kaiede, 2018-08-14:
Wrote up a quick driver for the 16-channel PCA9685 I2C PWM controller. Adafruit uses this on their PWM/Servo bonnet/HAT. 

https://github.com/Kaiede/PCA9685

uraimo, 2018-08-22:
Great! I know that board and had it in my TODO list, I'll update the readme right away, extremely useful for those who need a lot of PWMs.
(Yep, I've seen your other message too, sorry for the delay :) )