uraimo/SwiftyGPIO #17: Swift 3.0 Support

colemancda, 2016-06-05:
I would like to request Swift 3.0 support. If there is no progress made, then I will try porting it myself, but I would much rather it be a collaborative effort. Would you be interested?

uraimo, 2016-06-06:
Sure!
If you have already started feel free to post it here, we'll go on from there. I don't expect that the lib will need many changes, btw.

Regarding the use of a custom Thread class for that init(main:) that is not available on iOS/OSX and has a maybe uncertain future on other platforms, we could go either way, keep it as it is or depend on that portion of SwiftFoundation. Does someone have any opinion on this?

hpux735, 2016-06-06:
With all due respect to @colemancda and the amazing work of PureSwift, I'm pretty hesitant to start bifurcating things like Foundation. :/  But, I certainly wouldn't imply that I have any right to say what others should do! :)

colemancda, 2016-06-06:
I don't have any intention of making this project dependent on SwiftFoundation, but I would like to include Thread.swift

hpux735, 2016-06-06:
Fair enough :)

uraimo, 2016-06-06:
Ok, considering that Foundation is only needed for the NSThread class by the gpio listeners,  and that on some platforms a build with Foundation could not even be available, it makes sense.

PR #19 (2016-06-06): Update for Swift 3.0
Fixes #17
closed 2016-06-06 by uraimo