uraimo/SwiftyGPIO #30: Pull up/down support for RaspberryPi

uraimo, 2016-08-03:
Missing for now, it should be straightforward to implement.

uraimo, 2017-03-08:
Easy as expected but it looks like setting the pull status crashed an IRQ handler on the port... and after a while the RaspberryPi freezes completely.

The full source to set this parameter:

```swift
#if os(Linux)
import Glibc
#else
import Darwin.C
#endif

let BLOCK_SIZE = 4*1024

let BCM2708_PERI_BASE: Int = 0x3F000000
let GPIO_BASE: Int = BCM2708_PERI_BASE + 0x200000 /* GPIO controller */
let PWM_BASE: Int =  BCM2708_PERI_BASE + 0x20C000 /* PWM controller */
let CLOCK_BASE: Int = BCM2708_PERI_BASE + 0x101000 

var  mem_fd = open("/dev/mem", O_RDWR | O_SYNC)
if mem_fd <= 0 {
    fatalError("Can't open /dev/mem")
}

let gpio_map = mmap(
    nil,                 //Any adddress in our space will do
    BLOCK_SIZE,          //Map length
    PROT_READ|PROT_WRITE,// Enable reading & writting to mapped memory
    MAP_SHARED,          //Shared with other processes
    mem_fd,              //File to map
    off_t(GPIO_BASE)     //Offset to GPIO peripheral
    )!

let gpioBasePointer = gpio_map.assumingMemoryBound(to: UInt.self)
if (gpioBasePointer.pointee == UInt(bitPattern: -1)) {    //MAP_FAILED not available, but its value is (void*)-1
    print("mmap error: " + "\(gpioBasePointer)")
    abort()
}

close(mem_fd)    

enum Pull: UInt{
    case neither = 0
    case down    = 1
    case up      = 2
}

func gpioPull(_ value: Pull, id: UInt){
    //guard self.direction == .IN else {fatalError("Trying to set pull up/down on an output GPIO, abort!")}

    let gpioGPPUDPointer = gpioBasePointer.advanced(by: 25)
    gpioGPPUDPointer.pointee = value.rawValue   // Configure GPPUD
    usleep(10);                                 // 150 cycles or more
    let gpioGPPUDCLK0Pointer = gpioBasePointer.advanced(by: 26)
    gpioGPPUDCLK0Pointer.pointee = (1<<id)      // Configure GPPUDCLK0/1 for specific gpio
    usleep(10);                                 // 150 cycles or more
    gpioGPPUDPointer.pointee = 0                // Reset GPPUD
    usleep(10);                                 // 150 cycles or more
    gpioGPPUDCLK0Pointer.pointee = 0            // Reset GPPUDCLK0/1 for specific gpio
    usleep(10);                                 // 150 cycles or more
} 

gpioPull(.up, id:2)

print("Done!")
```

uraimo, 2017-09-28:
I'm adding this to [Hacktoberfest](https://hacktoberfest.digitalocean.com/) because even if it's not a starter task, some of you could find interesting debugging something that does not require just coding.
The simple code above should conform to Broadcom's specification but last time i tried it didn't work and never got around at fixing it.

You'll need a voltage tester for this 😉

Fishybob, 2017-12-16:
Hi Uraimo,

  I believe you are missing "0x" x2...
  (clobbering GPHEN0/GPHEN1 with hex word offsets written as decimal. ;-)

fix:
    let gpioGPPUDPointer = gpioBasePointer.advanced(by: 0x25)
    let gpioGPPUDCLK0Pointer = gpioBasePointer.advanced(by: 0x26)

  Thanks for making SwiftyGPIO!

Scott.

Verification:
pi@pi0w:~/swift/pulluptest $ raspi-gpio get 17
GPIO 17: level=0 fsel=0 func=INPUT
pi@pi0w:~/swift/pulluptest $ gpio mode 0 up
pi@pi0w:~/swift/pulluptest $ raspi-gpio get 17
GPIO 17: level=1 fsel=0 func=INPUT
pi@pi0w:~/swift/pulluptest $ sudo ./.build/debug/pulluptest
Hello, Linux!
gpioBasePointer: 0xb6f04000
gpioGPPUDPointer : 0xb6f04094
gpioGPPUDCLK0Pointer: 0xb6f04098
gpioPull(.down, id:17)
Done!
pi@pi0w:~/swift/pulluptest $ raspi-gpio get 17
GPIO 17: level=0 fsel=0 func=INPUT

uraimo, 2017-12-16:
Thank you for spotting this!! Stepping away from the keyboard would have helped I guess :D

uraimo, 2017-12-17:
Committed with 472cfd6ff218b64538eb106cbbc89f18d54e3ec1.

closed 2017-12-17 by uraimo