uraimo/SwiftyGPIO #44: Does not appear to work with the new Raspberry Pi Zero W

stulevine, 2017-03-24:
Is there something special that needs to happen in order to use SwiftyGPIO with the new Raspberry Pi Zero W?  I have swift 3.0.2 installed and working (I was able to build and run some existing apps that do not use SwiftyGPIO).  And, I am going by the RPi Zero GPIO Header chart and have an LED hooked up to GPIO #3. Also tested that it works with the command line gpio program, and was able to get the LED to power on/off by setting GPIO pin 3 to high/low.

I'm using the Examples/Basic/raspi1PlusAndZero/Sources/main.swift file (had to change the .RaspberryPiRevPlusZero to .RaspberryPiPlusZero)
But when I build and run the program (it does run since I put some debug print statements in to confirm) I don't see a flashing LED and I think I should.

uraimo, 2017-03-24:
It looks like you are the first one trying it on the W, afaik it shouldn't have anything different from an armv6 RaspberryPi 1 or Zero. I'm trying to see if they changed anything relevant in Raspbian.

uraimo, 2017-03-24:
Can you try with some other pin like .P18? (6th on the right side)

uraimo, 2017-03-24:
And if that doesn't work either, can you try the customgpio example (/Examples/CustomGPIO) using 3 or 18 as id?

These three lines should be enough:
```swift
var gp01 = GPIO(name: "P18",id: 18))
gp01.direction = .OUT
gp01.value = 1
```

stulevine, 2017-03-24:
Wait, pin 3 as in PHYSICAL PIN 3 or GPIO Pin 3?  Yes, physical pin 3 is i2c. However, I'm hooked up to GPIO Pin 3 which is physical pin 15.  When I use GPIOName in my source code, what does that represent?  Physical Pin or GPIO Pin?  Here is the header diagram I am referencing:

[image]

uraimo, 2017-03-24:
The library still uses the original numbering (that should still be the one you need to use with /sys/class/gpio/gpioX) described here: https://github.com/uraimo/SwiftyGPIO/wiki/GPIO-Pinout

So, in this case it looks like you were using gpio22 or .P22.

uraimo, 2017-03-24:
Ok, it seems that the one i'm using here is called Broadcom numbering scheme, while you were using the wiringPi numbering.

stulevine, 2017-03-24:
Yup, just posted right after I saw your post.  Thanks.  All is good.

uraimo, 2017-03-24:
Thank you! I've added a note in the readme to clarify this.

closed 2017-03-24 by uraimo