uraimo/SwiftyGPIO #53: fatal error: unexpectedly found nil while unwrapping an Optional value: file I2C.swift, line 350

0x6e3078, 2017-06-15:
I get an error while using the i2c.writeByte(HTU21DF_ADDRESS, value: HTU21DF_READHUM) command. 
I using the current master version of the Library. Is this a known issue?

uraimo, 2017-06-15:
It's a new one, thanks for opening an issue on this.

I guess that with 3.1.1 they are now enforcing everywhere the fact that UnsafePointers cannot contain null pointers because you should use optional pointers with that. What you see at line 350 was a workaround for checks introduced with 3.1.

Give me a few hours and I'll push a fix.

uraimo, 2017-06-15:
Fixed!

I noticed that you are using the i2c write that does not need to specify the register/command, if that does not work, try to address the register 0 using `i2c.writeByte(HTU21DF_ADDRESS,command: 0, value: HTU21DF_READHUM)`.

commit ac9f7772 (2017-06-15): Fixed null pointer issue for I2C

Closes #53.
closed 2017-06-15 by uraimo