uraimo/SwiftyGPIO #59: swift 4 support?

MasterSwift, 2017-12-28:


uraimo, 2017-12-29:
Adding support for Swift 4 will require just minimal changes, but I'm holding back until Swift 4 will be available on ARM. Right now the higher release you can run is 3.1.1.
See https://github.com/uraimo/buildSwiftOnARM/issues/5

helje5, 2018-06-05:
This is working now and can be closed. I have it running successfully in a 4.1 build:
```
docker run --rm --cap-add SYS_RAWIO --privileged --device /dev/mem \
           -it --name swiftfun helje5/rpi-swift-dev:4.1.0 /bin/bash
```

In the container:
```
swift@6526f76b21e2:~$ swift --version
Swift version 4.1 (swift-4.1-RELEASE)
Target: armv7-unknown-linux-gnueabihf

swift@6526f76b21e2:~$ mkdir testit
swift@6526f76b21e2:~$ cd testit/
swift@6526f76b21e2:~/testit$ swift package init --type executable
```

Package.swift:
```swift
// swift-tools-version:4.0
// The swift-tools-version declares the minimum version of Swift required to build this package.

import PackageDescription

let package = Package(
    name: "testit",
    dependencies: [
        // Dependencies declare other packages that this package depends on.
        .package(url: "https://github.com/uraimo/SwiftyGPIO.git", from: "0.0.0"),
    ],
    targets: [
        // Targets are the basic building blocks of a package. A target can define a module or a test suite.
        // Targets can depend on other targets in this package, and on products in packages which this package depends on.
        .target(
            name: "testit",
            dependencies: [ "SwiftyGPIO" ]),
    ]
)
```

Test program
```swift
import SwiftyGPIO
import Glibc

let gpios = SwiftyGPIO.GPIOs(for: .RaspberryPi3)

var button = gpios[.P22]!
print("Button:", button)

button.onChange  { gpio in
  print("Value changed:", gpio)
}

print("Current Status")
print("Direction: ", button.direction.rawValue)
//print("Edge:      ", button.edge.rawValue)                                                           
print("Active Low:", String(button.activeLow))
print("Value:     ", String(button.value))

print("Looping ...")

repeat {
  sleep(1)
}
while true
```

Run:
```
$ sudo .build/armv7-unknown-linux-gnueabihf/debug/testit
[sudo] password for swift:
Button: SwiftyGPIO.RaspberryGPIO
Current Status
Direction:  in
Active Low: true
Value:      0
Looping ...
Value changed: SwiftyGPIO.RaspberryGPIO 1
Value changed: SwiftyGPIO.RaspberryGPIO 0
```

uraimo, 2018-06-05:
Thanks for testing it, it's finally time for the 1.0.0 release I guess.

closed 2018-06-05 by uraimo