uraimo/SwiftyGPIO #71: Don't hard-crash when `/dev/mem` can't be opened

helje5, 2018-06-14:
Currently you quickly run into a
```
Fatal error: Can't open /dev/mem , use sudo!: file ../.build/checkouts/SwiftyGPIO.git-1203819502/Sources/SwiftyGPIO.swift, line 353
```
when a tool is started w/o sudo.

Since that is so common, I would rather provide a proper API for that (probably something throwing an Error). The (user)-dev should handle that case, and provide a clean exit.

uraimo, 2018-06-18:
Yeah, it makes sense when swifygpio is just used in a subsystem and the application should continue to work regardless of gpio failures.
I'll take a look at it, since GPIO objects are pre-initialized in the presets I'll likely turn those read/write methods into throwing methods and reorganize those checks.

MotivDev, 2019-05-08:
What is the status of this requested change?  I would like to see something other than abort in the I2C module as well.  I do not want to have my system taken down due to an I2C bus problem that impacts system functionality but it should not keep the remaining system components from operating.

Maybe it could return a Result type in its place which returns the data or an error depending on outcome?

uraimo, 2019-05-08:
I will do a new release tomorrow or so and the refactor of the error handling will be one of the changes included 👍

uraimo, 2019-05-12:
@MotivDev The [next_release](https://github.com/uraimo/SwiftyGPIO/tree/next_release) branch should already have what you need, for the original request of this issue I'll add and optional, and not really nice to look at, way to check if a GPIO/PWM is usable shortly.

Note: SPM should ignore the format of these -beta<n> tags and use the last 2.0.0-??? tag if you set that release in your descriptor (if not, it's a bug).