uraimo/SwiftyGPIO #72: Question about i2c

anohren, 2018-06-30:
https://github.com/uraimo/SwiftyGPIO/blob/c59c2ce8416137533cc0bd98ca8e4594fa09b990/Sources/I2C.swift#L335

Perhaps it's because I'm tired and it's very late but I couldn't get past this line when reading the code. What is the purpose of the right-shift? Doesn't it always result in zero?

uraimo, 2018-06-30:
No, you are right, regardless of what I was trying to do there, it's wrong anyway (and likely I've never used it in my libraries, the device I use are usually byte-based).

closed 2018-06-30 by uraimo

anohren, 2018-06-30:
Thanks. I was reading in order to determine whether it would be possible to talk to an LCD using i2c via PCF8574 (wiringpi implementation: https://git.drogon.net/?p=wiringPi;a=blob_plain;f=wiringPi/pcf8574.c;hb=HEAD) 

Could the current architecture support something like that? From what I understand it would require implementing a PCF8574-specific `GPIO`, so that they can be passed onto the LCD. Is the GPIO class at all suited for that?

It's my first contact with i2c and GPIO programming so it's not obvious to me what the purpose is of all its members, but I got the impression that `GPIO` is somewhat tailored towards the "native" GPIOs.

uraimo, 2018-07-01:
Yes, you could use the i2c to program the PCF8574 to output series of 8 bits to 8 parallel pins, see [this](https://github.com/uraimo/DS1307.swift/blob/master/Sources/DS1307.swift) for an example of how to use the I2C class. Notice that the PCF does just that, byte to parallel bits, it does not know anything about the devices you connect it to.

Alternatively you could connect the lcd (which one are you using?) directly to 8 gpios and program the lcd directly outputting one of the bits on each one. This will be slower and could not be really ok for graphic lcds were you plan to have > 10fps refresh rates.

anohren, 2018-07-01:
Sorry, I realize I was a bit unclear. I'm using a two row character lcd -- lcd1602 -- and was hoping to reuse your ´HD44780CharacterLCD.swift´. My question is whether the current `GPIO` class is suitable for customization so that it, using i2c, can represent the parallel output pins of the PCF, and then in turn can be passed to an instance of your `HD44780LCD` class? The LCD has an onboard PCF module so i2c is the only accessible interface.

In other words, I want to compose `HD44780LCD` with a currently non-existent `class PCF8574GPIO : GPIO` which handles the i2c. Are there any obvious problems with that idea?

I'm assuming your `HD44780LCD` implementation already does what I need since the 1602 can be used with the wiringpi lcd library, which itself claims to support HD44780U and the like (https://git.drogon.net/?p=wiringPi;a=blob;f=devLib/lcd.c;h=6c0e474948fded0be758e0cb8cb227afdf120189;hb=HEAD)

uraimo, 2018-07-01:
Oh ok, in that case I would recommend to just reimplement the [lcdWrite](https://github.com/uraimo/HD44780CharacterLCD.swift/blob/master/Sources/HD44780CharacterLCD.swift#L188) function to output everything via i2c to the PCF instead of what is currently being done, i.e. setting each bit on each one of the gpios.
I would not use the GPIO class at all, but instead replace all those d*n*.value in lcdWrite with just one `i2c.writeByte(<address>, command: <?>, value: <full value>)`.

Since the i2c chip is already build into the module, I'd take a look to the module docs to discover the board address and available command types. There could be only 1 or 2 commands/registers available (to read/write the HD44780 memory? I doubt they could have implemented something more complex) and if so you could be able to figure them out using `i2cdetect/i2cdump/i2cset` even if there is no documentation.

anohren, 2018-07-03:
Thanks for the pointers! I'll try and find answers to your questions as my next step and hopefully I'll gain some more insight. 

Right now all I know is that it came in a starter kit and that their instructions used wiringpi, which seemingly supported abstracting over components like the pcf. 

I was hoping to eventually contribute back something similar, like a pcf class, but taking your answer into account it sounds as though the current SwiftyGPIO architecture doesn't really support decoupling the PCF logic from the LCD logic. I think it would open up to contributions if it did, but then again this domain is new to me and perhaps there is no benefit to be had here.

uraimo, 2018-07-05:
For the way I've structured it, the library only provides access to the interfaces available on the Pi and both the PCF and LCD are seen as two distinct external devices. Following this approach, I'd build two additional device libraries, one for the PCF (using the i2c internally, this one doesn't do much) and one for the LCD that would just send stuff (lcd configuration) over to the PCF object. This is my interpretation of logic decoupling, the lcd logic (i.e. the bytes sent to the LCD like what is being done in the HD44780LCD lib) can be placed in the lcd obj and the PCF micro-library just sends those raw bytes to the devices.
It's easier than it looks, imo.

The fact is that there are a multitude of devices that can be connected to either the i2c or the spi and that alter in some way what happens to the output once it's sent out of the Pi and that sometime provide complex functionalities (e.g. i2c chips that provide multiple PWM output, spi chips that read/write analog values), and those are easier to integrate if provided as separate libraries every time you need them (the same thing you do with Arduino sketches when you import libs for specific devices).