uraimo/SwiftyGPIO #79: Errors on Raspberry Pi 3 B+

AdrianBinDC, 2018-12-26:
### Board Type

_Specify which kind of ARM board you are using, e.g. RaspberryPi3, RaspberryPi1, BBB, etc..._
Raspberry Pi 3 B+

### Operating System

_Which OS are you using? e.g. Ubuntu Mate 16.04, Raspbian, Armbian, etc..._
Hypriot 1.9.0

### Swift Version

_Which Swift version are you using? Did you compile it yourself or did you use prebuilt binaries?_
4.1.2 using [pre-built binaries](https://github.com/helje5/dockSwiftOnARM)

### Description

_Describe your hardware setup (e.g. board connected to i2c sensor) and describe briefly the issue, specifying the steps needed to reproduce it, if necessary._
I have an LED on the breadboard with a resistor. The ground is connected to P6, the positive is connected to P4 on a Raspberry Pi 3 B+

I've got a project created that compiles. The code in main.swift is as follows:

```
import Glibc
import SwiftyGPIO

var gp4 = GPIO(name: "P4", id: 4)
gp4.direction = .OUT

repeat {
    gp4.value = (gp4.value == 0) ? 1 : 0
    usleep(150 * 1000)
} while(true)
```

I also tried declaring `gp4` like this:

```
let gpios = SwiftyGPIO.GPIOs(for: .RaspberryPi3)
var gp4 = gpios[.P4]!
```

_Please list any error message you've receveived during execution._

When `gp4` is declared like this:

```
var gp4 = GPIO(name: "P4", id: 4)
gp4.direction = .OUT
```
> Fatal error: Unexpectedly found nil while unwrapping an Optional value: file /home/swift/GPIOTest/.build/checkouts/SwiftyGPIO.git-1189449763/Sources/SwiftyGPIO.swift, line 104
> Current stack trace:
> Illegal instruction (core dumped)
> 

When `gp4` is declared like this:

```
let gpios = SwiftyGPIO.GPIOs(for: .RaspberryPi3)
var gp4 = gpios[.P4]!
```

> Fatal error: Can't open /dev/mem , use sudo!: file /home/swift/GPIOTest/.build/checkouts/SwiftyGPIO.git-1189449763/Sources/SwiftyGPIO.swift, line 353
> Current stack trace:
> Illegal instruction (core dumped)

uraimo, 2018-12-26:
Hi Adrian, I've never tried Hypriot but it should work.
Where are you running this, directly in Hypriot or in a docker container ([gpio tutorial for docker](https://blog.hypriot.com/post/lets-get-physical/))?
Both samples require that you launch them with root privileges using `sudo`, and the first should work if `/sys/class/gpio/export` is available.
The second one should work too if they've imported the required kernel patches from Raspbian, but I'd say it's likely that they've done it.
Can you try again using sudo and maybe with [Swift 4.1.3 for Raspbian](https://www.dropbox.com/s/e257cvg23ghe2dt/swift-4.1.3-RPi23-RaspbianStretch.tgz?dl=0)?

uraimo, 2019-02-27:
Oh, and depending on the specific image, GPIO mapping could be disabled by default and could require manual activation, see: https://github.com/futurejones/swift-arm64/issues/2

> Adding "iomem=relaxed" to /boot/firmware/cmdline.txt is enough to get around this.