uraimo/SwiftyGPIO #80: Add mock classes for UARTInterface, SPIInterface, I2CInterface, ADCInterface, OneWireInterface, PWMOutput

haikieu, 2018-12-28:
### Board Type

Any

### Operating System

Any

### Swift Version

Any

### Description

Currently, the SwiftGPIO interfaces are tight up to concrete classes which heavily depend on Raspberry Pi's environment, OS and libraries. The downside is the code which has integrated SwiftyGPIO gets crash when run on MAC.

In order to take advanges of XCode, and enable debugging capable on MAC/XCode, I'd like to add mocks feature into SwiftyGPIO. 

The overall feature will conduct these things:
1) Implement mock classes of UARTInterface, SPIInterface, I2CInterface, ADCInterface, OneWireInterface, PWMOutput.
2) Mock classes provide extra closures, so that developers can provide any simulation logic into any certain function of mock object.
2) The concrete object shouldn't be created directly, but by help of InstanceCreator
3) InstanceCreate is able to instantiate the actual objects or alternative mock objects depending on 2 things, enableMock flag and the current OS. if enableMock is true and the OS is MAC then return the mock object, otherwise return the actual object

Check out this branch for more detail: https://github.com/haikieu/SwiftyGPIO/tree/Add-Mocks

uraimo, 2019-01-08:
Hi Hai, thanks for starting this, it was something I planned to add but never got around to it.
I'd like to request a few small modifications before merging it, can you open a PR?

Regarding the changes, the InstanceCreator decides which kind of concrete class should be instantiated depending on the platform/flag, hiding from the user details on the instance he's actually getting.
In my opinion it's better to make this logic explicit and visible, removing the InstanceCreator and adding a new "MockBoard" in `SupportedBoard` and return mocks in every instance creation function i.e. `GPIOs(for board:)`,`hardwareSPIs(for board:)`, etc... 

I like the idea of the configurable closures, it could make sense one day to provide basic defaults for testing (to be discussed).

haikieu, 2019-04-17:
Hi Uraimo,

Sorry for delay 3 months late. Your idea about MockBoard may be valid, I'll review my code again.

wlisac, 2020-01-23:
> In my opinion it's better to make this logic explicit and visible, removing the InstanceCreator and adding a new "MockBoard" in `SupportedBoard` and return mocks in every instance creation function i.e. `GPIOs(for board:)`,`hardwareSPIs(for board:)`, etc...

I really like the approach of a "MockBoard" – the ability to mock for testing would be 💯