uraimo/SwiftyGPIO #81: I2C write fails on "writeData" to ADC1115

robreuss, 2019-01-11:
### Board Type

RaspberryPi3 B

### Operating System

Ubuntu Mate 16.04

### Swift Version

Pre-built 4.1.3

### Description

Communicating with the [ADC1115](http://www.ti.com/lit/ds/symlink/ads1115.pdf) analog to digital convertor.  

I'm basing my implemented on the approach described [here](https://github.com/ControlEverythingCommunity/ADS1115/blob/master/Arduino/ADS1115.ino) for Arduino.

Communication with the device is fine.  `i2cdetect -y 1` reports the board as present at address 0x48.

Here is my code:

```swift
i2c.writeByte(72, value: 1)
i2c.writeByte(72, value: 132)
i2c.writeByte(72, value: 131)

let myData: [UInt8] = [1, 132, 131]
// PROBLEM IS HERE...
i2c.writeData(72, command: 0, values: myData)

usleep(3000)
while true == true {
    i2c.writeByte(72, value: 0)
    let byte1 = i2c.readByte(72)
    let byte2 = i2c.readByte(72)
    //let resData = i2c.readWord(72, command: 0)
    print(byte1, byte2)
    //print(resData)
    usleep(5000)
}
```
_Please list any error message you've receveived during execution._

When I attempt to do a `writeData` write, I get this error message:
```
true
I2C write failed: Input/output error
Aborted
```

(True refers to the result of `isReachable(72)`.)

robreuss, 2019-01-11:
BTW, the Arduino code I reference above works with the same chip using an Arduino with a variable resistor.

uraimo, 2019-01-25:
(I have seen this issue, need to find some time to look into this :) )

uraimo, 2019-02-20:
Looking at the arduino library, they write one byte at a time even if they could have used [Wire.write(data, length)](https://github.com/arduino/ArduinoCore-avr/blob/master/libraries/Wire/src/Wire.cpp#L218) that does the same thing, weird.

But in our case sending multiple bytes performs a `I2C_SMBUS_BLOCK_DATA` ioctl call, completely different from the basic `I2C_SMBUS_BYTE_DATA` that would have been performed sending a single byte. While now I don't remember the specifics of what happens from the protocol point of view, I'm nearly certain that if the specific I2C register does not support data blocks the call will fail.

Have you tried sending single bytes?

Also, and this is likely why this fails, this:
```
let myData: [UInt8] = [>>>>>1<<<<, 132, 131]
i2c.writeData(72, command: >>>>>>0<<<<<<<, values: myData)
```
does not match with the arduino code:
```
  // Start I2C Transmission
  Wire.beginTransmission(Addr);
  // Select configuration register <<<<<<<<<<<<<<<<<<<<<<<<<<<
  Wire.write(0x01); <<<<<<<<<<<<<<<<<<<<<<<<<<<<<<<
  // AINP = AIN0 and AINN = AIN1, +/- 2.048V
  Wire.write(0x84);
```

That `1` is the number of the configuration register(other call it "command" index) inside the device 0x48, this is how I would translate the arduino code:
```
i2c.writeByte(address, command: 0x1, value: 132) // To Reg0x1: AINP = AIN0 and AINN = AIN1, +/- 2.048V
i2c.writeByte(address, command: 0x1, value: 131) // To Reg0x1: Continuous conversion mode, 128 SPS
//This could work too, but doubt it:    let myData: [UInt8] = [132, 131]
//                                                            i2c.writeData(0x48, command: 0x1, values: myData)

usleep(3000)
while true == true {
    let byte1 = i2c.readByte(0x48, command: 0x0)  // This is how a register is selected, not with i2c.writeByte(72, value: 0)
    let byte2 = i2c.readByte(0x48, command: 0x0)
    //This could work too: let byte1 = i2c.readByte(0x48)
    //                                   let byte2 = i2c.readByte(0x48)
    //Or maybe this directly: let resData = i2c.readWord(0x48, command: 0)
    print(byte1, byte2)
    //print(resData)
    usleep(5000)
}
```

robreuss, 2019-02-20:
I'm guessing that I tried sending single bytes at some point - I was trying everything - but don't recall for sure.  When I get back to working with that chip I'll definitely give your code a try and report back in case it helps others.  Thanks!

dilames, 2021-12-23:
@uraimo 
Can you help me?
Your code seems to be working!
I'm quite confused how to correctly configure ASD1115 for reading from A0 input.

Can you help me?