uraimo/SwiftyGPIO #82: Trouble getting data off the LSM303 accelerometer/compass

robreuss, 2019-01-25:
### Board Type
RPi 3B

### Operating System

Ubuntu Mate 16.04

### Swift Version

Pre-built 4.1.3

### Description

Trying to communicate with an LSM303 but getting mostly zeros.  I don't think this is a problem with the library but I could use guidance on when to use `readData` versus `readByte` if you could!

I'm porting from Python:
https://github.com/adafruit/Adafruit_Python_LSM303/blob/master/Adafruit_LSM303/LSM303.py#L85

I'm enabling the magnetometer based on this **Python** command:
```python
self._mag.write8(LSM303_REGISTER_MAG_MR_REG_M, 0x00)
```

I'm issuing the following **SwiftyGPIO** command for that, to enable the mag:
```swift
i2c!.writeByte(magAddress, command: UInt8(LSM303_REGISTER_MAG_MR_REG_M), value: 0x00)
```

I'm port the following **Python** command to read data:
```python
mag_raw = self._mag.readList(LSM303_REGISTER_MAG_OUT_X_H_M, 6)
```

I'm issuing the following **SwiftyGPIO** command for that, to read data:
```swift
let gyroData = i2c!.readData(magAddress, command: UInt8(LSM303_REGISTER_MAG_OUT_X_H_M)
```

Should I be fetching six bytes using `readByte` instead?  I think not.

I'm getting the following data in response:
```swift
[104, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
```
The first six bytes should be meaningful, with hi/lo bytes representing the magnetometer raw data.

I've tried a lot of permutations but I'm confident I've nailed it with this approach.

Thanks!

closed 2019-02-19 by robreuss

uraimo, 2019-02-20:
Did it work in the end?

robreuss, 2019-02-20:
I found this driver and didn't dig too deep to find out what I was doing wrong.  The driver has good core functionality so perhaps it should be added to your list of projects!
https://github.com/flx/LSM303

uraimo, 2019-02-20:
Oh nice! Thanks!