uraimo/SwiftyGPIO #84: Cannot use two PWM channels at the same time

robreuss, 2019-03-01:
### Board Type

RaspberryPi3

### Operating System

Ubuntu Mate 16.04

### Swift Version

4.1.3, pre-built

### Description

Attempting to drive a motor controller (https://www.pololu.com/product/3754) using PWM.  I have one channel each for the right and left sides of my robot vehicle.  When I `startPWM` to one channel, the other channel stops.  This occurs even if the period and duty are the same with each `startPWM` call.  Any way to overcome this behavior?

uraimo, 2019-03-01:
Yes, because right now the clock that generates the pwm is being killed every time a new pwm is started, see [here](https://github.com/uraimo/SwiftyGPIO/blob/master/Sources/PWM.swift#L164) (I was more interested in building something to easily generate patterns than in the vanilla pwm, so a few basic thing were never implemented).

If the frequency/ns parameter is kept the same (the duty cycle doesn't matter) this should not be needed, so it could be either removed, or better performed only the first time startPWM is called checking a flag.
To try this out you can simply modify the local checkout of this library (under .build/...) in your project.
Let me know if it doesn't work, but I'm somewhat confident it should :)

This was discussed with Kaiede some time ago in #73, and in the end in his improved fork he cleaned up the implementation reducing the number of moving parts and removing the need for the clock reset.

robreuss, 2019-03-01:
Cool.  Swapped in Kaiede's version and it works fine.  

I used an online calculator to convert my desired frequency from 20Khz to 50000 nanoseconds but I wanted to confirm with you that that is the correct nanosecond value for that frequency.

uraimo, 2019-03-01:
Yep, correct, 1Hz=1/1sec, so 1/20000Hz gives you 50 microseconds or 50000 nanoseconds.

Btw, since this could be a common annoyance for a lot of people I plan to fix this (and a few other things) and make a new release.

robreuss, 2019-03-01:
I'll watch for the new release!  Thanks!

closed 2019-03-01 by robreuss

robreuss, 2022-01-07:
I was wondering if this improvement was eventually made to SwiftyGPIO.  I've run into the same problem using SwiftyGPIO so I'm assuming not, but wanted to confirm before I switch over to SingleBoard.