uraimo/SwiftyGPIO #88: I2C Device Not Detected (SHT20)

samco182, 2019-05-30:
### Board Type

Raspberry Pi 3B+
### Operating System

Raspbian

### Swift Version

Swift 4.2.3, prebuilt binary from https://github.com/uraimo/buildSwiftOnARM/releases/tag/4.2.3

### Description
I have a SHT20 I2C sensor that I am trying to connect and use with the help of this library. According to the sensors data sheet the default address is **0x40** which es exactly what I get when I run:
```
sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: 40 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```
With this, I at least know that the device is connected correctly and not a connection problem.

But now, when running the following code (extracted from the example in the library), I don't get anything back, and even checking directly for `i2c.isReachable(0x40)` returns false, so I can't get it to work 😢 
```
import SwiftyGPIO

let i2cs = SwiftyGPIO.hardwareI2Cs(for:.RaspberryPi3)!
let i2c = i2cs[1]

print("Detecting devices on the I2C bus:\n")
outer: for i in 0x0...0x7 {
    if i == 0 {
        print("    0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f")
    }
    for j in 0x0...0xf {
        if j == 0 {
            print(String(format:"%x0",i), terminator: "")
        }
        // Test within allowed range 0x3...0x77
        if (i==0) && (j<3) {print("   ", terminator: "");continue}
        if (i>=7) && (j>=7) {break outer}
        
        print(" \(i2c.isReachable(i<<4 + j) ? " x" : " ." )", terminator: "")
    }
    print()
}
print("\n")
print("Is Reachable: \(i2c.isReachable(0x40))")
```

This is what I get when running the code above:
```
➜  Example git:(master) ✗ swift run
Compile Swift Module 'SwiftySHT20Example' (1 sources)
Linking ./.build/armv7-unknown-linux-gnueabihf/debug/SwiftySHT20Example
Detecting devices on the I2C bus:

    0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00           .  .  .  .  .  .  .  .  .  .  .  .  .
10  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .
20  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .
30  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .
40  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .
50  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .
60  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .
70  .  .  .  .  .  .  .

Is Reachable: false
```

Am I missing some kind of configuration? Or does it seem like a bug?

uraimo, 2019-05-30:
Can you repeat the test with `SwiftySHT20Example` a few times spacing each request by 5 seconds?

(See next comment for more)

uraimo, 2019-05-30:
Ok, you can ignore the result of isReachable for now, the rest of the read/write commands (read/write byte in your case, check out the [https://github.com/DFRobot/DFRobot_SHT20/blob/master/DFRobot_SHT20.cpp](arduino library)) should work.

The problem here is that the SHT behaves exactly how `i2cdetect` expects, being:
```
       As  there  is  no  standard I2C detection command, i2cdetect uses arbitrary SMBus commands
       (namely SMBus quick write and SMBus receive byte) to probe for devices.  By  default,  the
       command  used is the one believed to be the safest for each address.
```
That *believed to be the safest* actually means:
```
Address:
0x03 --> 0x2F: write
0x30 --> 0x37: receive
0x38 --> 0x4F: write
0x50 --> 0x5F: receive
0x60 --> 0x77: write
```
The SHT follows exactly these rules allowing only writes for 0x48, but `isReachable` as it's implemented right now performs only a byte read.
Since other device manufacturer could follow what i2cdetect dictates (and there is no point in providing a different result) I'll update the reachable function  🤷‍♂️.

uraimo, 2019-05-30:
Added [i2c debugging guide](https://github.com/uraimo/SwiftyGPIO/blob/master/docs/i2c-debugging.md)

uraimo, 2019-05-30:
I've pushed 1.1.3 with the new `isReachable`, can you verify that the example above now works? Thanks.

samco182, 2019-05-30:
I tried to follow the debugging guide you added above, and got the following result:
```
       i2cdetect-1369  [003] ....   123.905023: i2c_write: i2c-1 #0 a=040 f=0000 l=0 []
       i2cdetect-1369  [003] ....   123.905234: i2c_result: i2c-1 n=1 ret=1
```
That was the only address that got back a `ret=1` the rest of them got a `ret=-121-`

Next, I updated the library to `1.1.3` to include your change to `isReachable` but it is still not working, I keep getting `false` whenever I ask for the reachability of address `0x40`:
```
➜  Example git:(master) ✗ swift run
Compile Swift Module 'SwiftySHT20Example' (1 sources)
Linking ./.build/armv7-unknown-linux-gnueabihf/debug/SwiftySHT20Example
Detecting devices on the I2C bus:

    0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00           .  .  .  .  .  .  .  .  .  .  .  .  .
10  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .
20  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .
30  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .
40  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .
50  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .
60  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .  .
70  .  .  .  .  .  .  .

Is Reachable: false
```
I even tried wrapping the code above in a loop to see if it was matter or time getting it reachable, but no luck. I left the code running a while, but all of the results were false.

uraimo, 2019-05-30:
> That was the only address that got back a ret=1 the rest of them got a ret=-121

With i2cdetect? And when you run the Swift example what do you get for 0x40?

samco182, 2019-05-30:
With i2cdetect I kept getting the value above :
```
sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: 40 -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
```
Ok, I am having some weird behavior. While i2cdetect always returns the print above, I am getting the Swift code to work sometimes only. There was a couple a times I got `isReachable: true` for the code above, I even got readings for:
```
print("I2C + Write: \(i2c.writeByte(0x40, value: 0xE7))")
print("I2C + Read: \(i2c.readByte(0x40))")
```
but after trying a couple of times it started returning the same as before (false for isReachable) and I am getting 
`I2C write failed: Remote I/O error`.

uraimo, 2019-05-30:
If you are trying to read_user_reg, you should read with `i2c.readByte(0x40, command:1)` I think (this could be the cause of the i/o error).
But you can just read the temperature/humidity:
```
i2c.writeByte(0x40, value: 0xF3) //TRIGGER_TEMP_MEASURE_NOHOLD
usleep(500_000) //import Glibc
let b0 = i2c.readByte(0x40)
let b1 = i2c.readByte(0x40)
let tmp = b0 << 8 + b1
let cTemp= -46.85 + ((Float(tmp) * 175.72) / 65536)
let fTemp = cTemp * 1.8 + 32

i2c.writeByte(0x40, value: 0xF5) //TRIGGER_HUMD_MEASURE_NOHOLD
usleep(500_000)
let b0 = i2c.readByte(0x40)
let b1 = i2c.readByte(0x40)
let tmp = b0 << 8 + b1
humidity = -6 + ((Float(tmp) * 125.0) / 65536.0)

print("Temperature: \(cTemp)°C/\(fTemp)°F, Relative Humidity: \(humidity)%")
```

Regarding the fact that sometimes it's reachable and sometimes it's not, can you check that the log output when you run the swift example (without the additional i2c write/read, that could make the device unreachable until it resets) is like the one you get from i2cdetect? (all ret=-121 except 0x40 that returns 1).

samco182, 2019-05-30:
Sorry, I didn't explain myself well. Forgot to mention that using `i2c.readByte(0x40, command:1)` and setting the command to either 1 o 0, always gives me the `I2C read failed: Remote I/O error`. The only method that works for me to read is `i2c.readByte(0x40)`, not sure why. So, that is the one I have used to read the user reg.

Thanks so much for the code , I tested it and it works! 🎉

Yes, I ran the I2C Swift scan code from above (without the additional i2c write/read) and this is what got recorded in log, just `0x40` gets `ret=1`:
```
 SwiftySHT20Exam-2219  [001] ....   269.735800: i2c_read: i2c-1 #0 a=040 f=0001 l=1
 SwiftySHT20Exam-2219  [001] ....   269.736021: i2c_reply: i2c-1 #0 a=040 f=0001 l=1 [3a]
 SwiftySHT20Exam-2219  [001] ....   269.736022: i2c_result: i2c-1 n=1 ret=1
...
 SwiftySHT20Exam-2219  [001] ....   269.747620: i2c_read: i2c-1 #0 a=040 f=0001 l=1
 SwiftySHT20Exam-2219  [001] ....   269.747838: i2c_reply: i2c-1 #0 a=040 f=0001 l=1 [3a]
 SwiftySHT20Exam-2219  [001] ....   269.747839: i2c_result: i2c-1 n=1 ret=1
```
I think you might be right about the additional i2c write/read code. After I added that code, and ran the Swift program again, it started failing, and stayed failing until I rebooted the Rasp. Worth noting that this **only happens** if I use 
```
print("I2C + Write: \(i2c.writeByte(0x40, command: 1, value: 0xE7))")
print("I2C + Read: \(i2c.readByte(0x40, command: 1))")
```
If use this instead, everything keeps working like normal
```
print("I2C + Write: \(i2c.writeByte(0x40, value: 0xE7))")
print("I2C + Read: \(i2c.readByte(0x40,))")
```

uraimo, 2019-05-31:
> always gives me the `I2C read failed: Remote I/O error`.

On some sensors doing something like `i2c.readByte(0x40, command:0xE7)` directly without requesting a specific function first works (even if not clearly stated in the docs).

> After I added that code, and ran the Swift program again, it started failing

Can I consider the original issue (scanning not working) fixed then?

Oh, and let me know if you ever turn the part related to the SHT in a standalone library, so that I'll add it to the rest in the readme :) .

samco182, 2019-05-31:
> On some sensors doing something like i2c.readByte(0x40, command:0xE7) directly without requesting a specific function first works (even if not clearly stated in the docs).
I will try that out as well ☝️ 

Yes, the original issue was resolved, thanks for the help! 🎖 

Yes, I am actually working on the standalone library right know! 🙌  That would be so great! I will for sure let you know once it is completed! :) (btw..where do I contact you? Email? Twitter?)

uraimo, 2019-05-31:
Great! Closing this then. 
Email or twitter are both ok 👍

closed 2019-05-31 by uraimo