uraimo/SwiftyGPIO #91: Swift 5 Support?

jhoughjr, 2019-07-02:
### Board Type

RPi 3 B+
### Operating System

Raspian Buster 

### Swift Version

prebuilt swift 5.0.1v0.3 from buildSwiftOnArm

### Description

building a small project that depends on MicroExpress and SwiftyGPIO

The build error is 

```
/home/pi/workspace/mod_fluid_control/.build/checkouts/SwiftyGPIO/Sources/Mailbox.swift:179:42: error: use of unresolved identifier 'gnu_dev_makedev'
        if mknod(filename, S_IFCHR|0600, gnu_dev_makedev(100, 0)) < 0 {
                                         ^~~~~~~~~~~~~~~
```

Some googling indicates it is part of glibc? I'm not sure.

uraimo, 2019-07-02:
Hi! Thanks for pointing this out, I saw it yesterday while compiling 5.1 but I now realise it's an issue with 5.x running on Buster.
Yep, that is part of Glibc and it should be there, but incredibly it's not.
I'll look into it today.

jhoughjr, 2019-07-02:
Is there any work around I can use in the meantime? I've tried locating the dependency to no avail.

uraimo, 2019-07-02:
Nope, Debian/Raspbian Buster has updated the libc header `sysmacros.h` that contains `gnu_dev_makedev`, and from the looks of it Swift shouldn't be able to import the new definition automatically.

Right now I'm reimplementing it but I'll check later if I can fix it directly in Swift.

uraimo, 2019-07-02:
Fixed, the new tag is `1.1.8`.

closed 2019-07-02 by uraimo

uraimo, 2019-07-02:
(1.1.9)

uraimo, 2019-07-02:
@jhoughjr  I don't know what you plan to do, but could you report back if you notice any issues?
I'm going through the bug report for Raspbian Buster and A LOT of things doesn't seem to work (issues on I2C, UART and even simple GPIOs).

jhoughjr, 2019-07-05:
Sorry I've got pulled away from the project that used this for a few days so I haven't tested it yet. Sometime this weekend I'll get back on it and can let you know.

uraimo, 2019-07-05:
No problem 👍

jhoughjr, 2019-07-11:
I'm finally back to messing with this so I should know if it works soon.