uraimo/SwiftyGPIO #93: Fatal error: Parameter cannot be read

cansurmeli, 2019-08-10:
### Board Type

Raspberry Pi Model 3B v1.2

### Operating System

Raspbian Buster

(everything up-to-date)

### Swift Version

Swift 5.0.2 with the pre-built binaries

### Description

I'm just trying to execute

`sudo ./.build/debug/Projects `

the sample code on the README:

```
import SwiftyGPIO

let gpios = SwiftyGPIO.GPIOs(for:.RaspberryPi3)
print(gpios)
var gp = gpios[.P2]!

print(gp)
```

 and I'm getting the following error:

```
Fatal error: Parameter cannot be read.: file /home/archie/Swift/Projects/.build/checkouts/SwiftyGPIO/Sources/SwiftyGPIO.swift, line 334
Current stack trace:
[1]    21475 illegal hardware instruction  sudo ./.build/debug/Projects
```

However, I don't get why I'm facing this error.

Any ideas? Thanks.

uraimo, 2019-08-16:
Hi! Sorry for the late reply, can you try this instead?

```swift
import SwiftyGPIO

let gpios = SwiftyGPIO.GPIOs(for:.RaspberryPi3)
print(gpios)
var gp = gpios[.P2]!
print("Current Status")
print("Direction: "+gp.direction.rawValue)
print("Edge: "+gp.edge.rawValue)
print("Active Low: "+String(gp.activeLow))
print("Value: "+String(gp.value))
gp.direction = .OUT
gp.value = 1
print("New Status")
print("Direction: "+gp.direction.rawValue)
print("Edge: "+gp.edge.rawValue)
print("Active Low: "+String(gp.activeLow))
print("Value: "+String(gp.value))
```

(Also, I've noticed that the CustomStringConvertible is trying to print .pull, it shouldn't since it's read-only)

uraimo, 2019-08-16:
Fixed with `1.1.10` removing the write-only .pull, let me know if there are other issues and thanks for opening this one.

closed 2019-08-16 by uraimo

cansurmeli, 2019-08-24:
Good to know. However, I'm currently not at home and traveling. I'll try this when I get back.

cansurmeli, 2019-09-26:
Just to let you know, I finally had a chance to try this with `1.1.10` which worked successfully.

Thanks.

uraimo, 2019-09-27:
Thanks for reporting back! 👍