uraimo/SwiftyGPIO #96: GPIO Inputs fail to execute `onChange`, `onFalling`, and `on Raising` closures

samco182, 2019-10-19:
### Board Type

Raspberry Pi 3B+

### Operating System

Raspbian 

### Swift Version

Swift 5.0.2, prebuilt binary from https://github.com/uraimo/buildSwiftOnARM/releases/tag/5.0.2

### Description
I have connected a push button to the Raspberry Pi with the following schematic:
[image]
(this image was just for reference, I have connected my push button to **PIN17** instead of PIN10)

The code setup is the following: 
```
import Foundation
import SwiftyGPIO

let gpios = SwiftyGPIO.GPIOs(for: .RaspberryPi3)
let realGPIO = gpios[.P17]!
realGPIO.direction = .IN
realGPIO.pull = .down
realGPIO.bounceTime = 0.5

print("GPIO STATE: \(realGPIO.value)")

realGPIO.onFalling { _ in
    print("Falling activated")
}

RunLoop.main.run()
```

Now, here is the problem: If I reboot my rasp, and immediately after reboot, I run my program, the `onFailling` closure (or any of the other available closures) never gets executed.  Now, if I quit the program, and run it again (without rebooting) then everything works fine, the closure is correctly called after each push. 
And this behavior happens **every time I reboot the rasp**, during the first program run the closure never works, and from the second and on, it works perfectly.

I've tried to follow the code to check where it gets stuck, and I've found out that **only after reboot** the code will get stuck on `SwiftyGPIO.swift`, `makeInterruptThread()` function on line `247`. It never gets out of the `poll` function call. From the second program run to `n` run, then it does not get stuck in `poll` function since it always returns a `1` on button push.

uraimo, 2019-10-19:
I have a few ideas on what could be wrong, I'll try them and report back.

uraimo, 2019-10-20:
It should work now, hard to tell what's happening underneath.
Opening the file as RW is enough to fix it, in my first attempts I've also noticed that doing a dummy poll for POLLIN|POLLOUT (events that are always true) works to when opened as read-only.

closed 2019-10-20 by uraimo

samco182, 2019-10-21:
@uraimo could you please tell me the code, and connections you used to test this? I am still having the same issue described above, even with the fix you pushed.
Or maybe I am missing some configuration or something 🤔

uraimo, 2019-10-21:
Yep, verified it's still there with a Pi3, works on a PiZero but I'm not really sure if the Zero completely reboots like the other board...

uraimo, 2019-10-21:
Can you check if the new workaround in 1.1.13 works for you too?

samco182, 2019-10-21:
Yes, it did work @uraimo! thanks for the help!

uraimo, 2019-10-21:
I knew the day when I'd have to fix something introducing some random sleep() would come :D  

In this case, the problem is not related to the `poll` itself but it has to do with the gpio export of `/sys/glass/gpio/gpio<nn>` that happens right before that.  Exporting it manually or with a program right after boot and then manually launching your sample would have worked too for example.
Hard to say what has to "settle down" on that file descriptor before we can start polling it.

samco182, 2019-10-21:
Our good ol' reliable `sleep()` saving the day again haha. Thanks for the help @uraimo!

commit ad82d795 (2019-10-20): Fixed issue with poll on sysfs /sys/class/gpio value on boot, closes #96 

Opening the fp as RW fixes the issue of poll blocking forever the first time a program is launched after boot.
Alternatively, do a dummy poll with POLLIN|POLLOUT (for example just before the dummy read).
commit 8b78a143 (2019-10-20): Fixed issue with poll on sysfs /sys/class/gpio value on boot, closes #96

Opening the fp as RW fixes the issue of poll blocking forever the first time a program is launched after boot.
Alternatively, do a dummy poll with POLLIN|POLLOUT (for example just before the dummy read).