uraimo/SwiftyGPIO #98: Failing to recognize UARTs on Raspberry Pi 3

rchatham, 2019-10-30:
### Board Type

Raspberry Pi 3, planning to test on Raspberry Pi Zero W.

### Operating System

Raspian

### Swift Version

Swift 5.1

### Description

Raspberry Pi 3 with 5in XPT2046 Touch Controller and a usb cable plugged in between the Pi and an Arduino Nano. The Arduino Nano is running pretested code reading values from a circuit and sending those values over its Serial connection to the Pi.

I received the following error:
```
Fatal error: Unexpectedly found nil while unwrapping an Optional value: file /home/pi/Swish/pi.reidchatham.com/.build/checkouts/SwiftyGPIO/Sources/UART.swift, line 46
```

rchatham, 2019-10-30:
It looks to me that it is failing to initialize the UARTs for the board. Any help would be greatly appreciated!

rchatham, 2019-10-30:
When running on Raspberry Pi Zero W I get the same error mentioned here as well so this is likely related. https://github.com/uraimo/SwiftyGPIO/issues/54#issuecomment-500077089

uraimo, 2019-10-31:
Can you double check if you have manually enabled the UART ports?
 
Run `sudo raspi-config` and then select "Interfacing Options", "Login shell via serial: NO", "serial hardware port enabled: YES".

rchatham, 2019-10-31:
No luck from what I can tell on the Raspberry Pi Zero W. I am using a MakerSpot Cocker Hub Board to get access to standard USB 2.0 ports rather than the micro-usb on the Pi Zero, not sure if that is complicating things. I typed `ls /dev/tty*` in into the terminal and got the following response:
```
pi@raspberrypi:~ $ ls /dev/tty*
/dev/tty    /dev/tty19  /dev/tty3   /dev/tty40  /dev/tty51  /dev/tty62
/dev/tty0   /dev/tty2   /dev/tty30  /dev/tty41  /dev/tty52  /dev/tty63
/dev/tty1   /dev/tty20  /dev/tty31  /dev/tty42  /dev/tty53  /dev/tty7
/dev/tty10  /dev/tty21  /dev/tty32  /dev/tty43  /dev/tty54  /dev/tty8
/dev/tty11  /dev/tty22  /dev/tty33  /dev/tty44  /dev/tty55  /dev/tty9
/dev/tty12  /dev/tty23  /dev/tty34  /dev/tty45  /dev/tty56  /dev/ttyAMA0
/dev/tty13  /dev/tty24  /dev/tty35  /dev/tty46  /dev/tty57  /dev/ttyS0
/dev/tty14  /dev/tty25  /dev/tty36  /dev/tty47  /dev/tty58  /dev/ttyUSB0
/dev/tty15  /dev/tty26  /dev/tty37  /dev/tty48  /dev/tty59  /dev/ttyprintk
/dev/tty16  /dev/tty27  /dev/tty38  /dev/tty49  /dev/tty6
/dev/tty17  /dev/tty28  /dev/tty39  /dev/tty5   /dev/tty60
/dev/tty18  /dev/tty29  /dev/tty4   /dev/tty50  /dev/tty61
```

rchatham, 2019-10-31:
I'm wondering if this library couldn't be used to interpret a serial connection via USB similar to using  [pyserial](https://github.com/pyserial/pyserial) as in the following tutorials ([1](https://www.instructables.com/id/Raspberry-Pi-Arduino-Serial-Communication/), [2](https://classes.engineering.wustl.edu/ese205/core/index.php?title=Serial_Communication_between_Raspberry_Pi_%26_Arduino)). This would seem to be a simple alternative way to make a serial connection between an Arduino and Raspberry Pi then using the GPIO and UART pins.

rchatham, 2019-10-31:
Running `ls -l /dev` in the terminal I see the following ports being forwarded.
```
lrwxrwxrwx 1 root root           5 Oct 31 17:24 serial0 -> ttyS0
lrwxrwxrwx 1 root root           7 Oct 31 17:24 serial1 -> ttyAMA0
```

uraimo, 2019-10-31:
> No luck from what I can tell on the Raspberry Pi Zero W.

The hub shouldn't be a problem, but even after enabling the uart (that is really there, looking at the ls printout) you still get the same error?

>  this library couldn't be used to interpret a serial connection via USB similar to

If you mean that you want to use that uart adapter connected to usb, sure:
```
let uart = SysFSUART("ttyUSB0")
```

uraimo, 2019-10-31:
I've just read again your first comment on the Zero W, can you paste the whole error?

rchatham, 2019-10-31:
Of course.

```
pi@raspberrypi:~ $ swift run --skip-build Run --hostname raspberrypi.local --port 8080
Couldn't open UART device: Permission denied
Fatal error: Index out of range: file /home/buildSwiftOnARM/swift/stdlib/public/core/ContiguousArrayBuffer.swift, line 369
```

rchatham, 2019-10-31:
> > this library couldn't be used to interpret a serial connection via USB similar to
> 
> If you mean that you want to use that uart adapter connected to usb, sure:
> 
> ```
> let uart = SysFSUART("ttyUSB0")!
> ```
Same issue as above if I try to use the usb input.

uraimo, 2019-11-01:
On the Zero (`.RaspberryPiPlusZero`), right now you have only one UART, either "serial0" or "ttyAMA0" , the one that is found first, so change that `uarts[1]` in `uarts[0]`.
In reality, the Zero has two, so I need to update the library splitting the Zero from the configurations needed for the Pi 1 (#99).

At the moment, if you need to use "ttyS0" just initialize it manually, but remember that S0 should refer to the UART for the serial linux console connection (aka terminal), I doubt that you need to connect to that one.  So change to 0 or to your custom ttyUSB0 in this case too.

The `Permission denied` error could depend from the port requiring higher privileges, it could be the case with ttyS0 considering what it does. `ttyUSB0` could have the same problem, try to run your program with sudo.

I've actually found an additional issue on readLine that I'm fixing but it shouldn't affect you case.

rchatham, 2019-11-07:
Hmmm strange. I got the `Permission denied` error on the Zero for `uarts[0]`. USB input seems to work without returning the `Permission denied` error but I haven't been able to get the data from it. I've migrated to using [SwiftSerial](https://github.com/yeokm1/SwiftSerial) though for my purposes. Thank you for the help and responsiveness @uraimo!

closed 2019-11-07 by rchatham

uraimo, 2019-11-08:
I wasn't able to replicate your issue but I'll look into it again when I'll start working on a few new device libraries.