uraimo/SwiftyGPIO #104: Iterating GPIOs crashes on RPi 4

KevinVitale, 2020-06-05:
### Board Type

 - RaspberryPi3B+,
 - RaspberryPi4 8GB

### Operating System

Ubuntu Server 18.04

### Swift Version

[Swift 5.2.4 v4](https://packagecloud.io/swift-arm/release/packages/ubuntu/focal/swift-lang_5.2.4-4-ubuntu-focal_arm64.deb)

### Description

Wrote a program to iterate over GPIOs:
```swift
import SwiftyGPIO

let gpios = SwiftyGPIO.GPIOs(for: .RaspberryPi3)
gpios
  .compactMap({ $0 })
  .forEach { print($0) }
```

When I run this on RPi 3B+, it works without a problem:
```
➜  PiTest sudo swift run
(key: SwiftyGPIO.GPIOName.P26, value: GPIO26<OUT,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P21, value: GPIO21<IN,NONE,true>: 1)
(key: SwiftyGPIO.GPIOName.P10, value: GPIO10<OUT,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P23, value: GPIO23<IN,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P12, value: GPIO12<IN,NONE,true>: 1)
(key: SwiftyGPIO.GPIOName.P15, value: GPIO15<OUT,NONE,true>: 1)
(key: SwiftyGPIO.GPIOName.P19, value: GPIO19<OUT,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P14, value: GPIO14<OUT,NONE,true>: 1)
(key: SwiftyGPIO.GPIOName.P7, value: GPIO7<OUT,NONE,true>: 1)
(key: SwiftyGPIO.GPIOName.P6, value: GPIO6<OUT,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P11, value: GPIO11<OUT,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P5, value: GPIO5<IN,NONE,true>: 1)
(key: SwiftyGPIO.GPIOName.P18, value: GPIO18<IN,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P9, value: GPIO9<OUT,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P27, value: GPIO27<IN,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P3, value: GPIO3<OUT,NONE,true>: 1)
(key: SwiftyGPIO.GPIOName.P17, value: GPIO17<IN,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P22, value: GPIO22<IN,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P24, value: GPIO24<IN,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P20, value: GPIO20<IN,NONE,true>: 1)
(key: SwiftyGPIO.GPIOName.P4, value: GPIO4<IN,NONE,true>: 1)
(key: SwiftyGPIO.GPIOName.P8, value: GPIO8<OUT,NONE,true>: 1)
(key: SwiftyGPIO.GPIOName.P25, value: GPIO25<IN,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P13, value: GPIO13<OUT,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P16, value: GPIO16<IN,NONE,true>: 1)
(key: SwiftyGPIO.GPIOName.P2, value: GPIO2<OUT,NONE,true>: 1)
```

When I run this on RPi 4, it iterates a few, then crashes with a `nil` exception:

```
➜  PiTest sudo swift run
(key: SwiftyGPIO.GPIOName.P25, value: GPIO25<IN,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P13, value: GPIO13<IN,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P18, value: GPIO18<IN,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P27, value: GPIO27<IN,NONE,true>: 0)
Fatal error: Unexpectedly found nil while unwrapping an Optional value: file /home/ubuntu/Swift/PiTest/.build/checkouts/SwiftyGPIO/Sources/SwiftyGPIO.swift, line 73
Current stack trace:
0    libswiftCore.so                    0x0000ffff7f9cce4c swift_reportError + 64
1    libswiftCore.so                    0x0000ffff7fa320e4 _swift_stdlib_reportFatalErrorInFile + 128
2    libswiftCore.so                    0x0000ffff7f7867b0 <unavailable> + 1439664
3    libswiftCore.so                    0x0000ffff7f785a28 _assertionFailure(_:_:file:line:flags:) + 464
4    PiTest                             0x0000aaaadd95ee64 <unavailable> + 343652
5    PiTest                             0x0000aaaadd96363c <unavailable> + 362044
6    PiTest                             0x0000aaaadd9637b4 <unavailable> + 362420
7    libswiftCore.so                    0x0000ffff7f7cb5e0 _debugPrint_unlocked<A, B>(_:_:) + 304
8    libswiftCore.so                    0x0000ffff7f835120 <unavailable> + 2154784
9    libswiftCore.so                    0x0000ffff7f7cacb4 _print_unlocked<A, B>(_:_:) + 740
10   libswiftCore.so                    0x0000ffff7f83b5bc <unavailable> + 2180540
11   libswiftCore.so                    0x0000ffff7f83a680 <unavailable> + 2176640
12   PiTest                             0x0000aaaadd915fc8 <unavailable> + 45000
13   PiTest                             0x0000aaaadd916068 <unavailable> + 45160
14   PiTest                             0x0000aaaadd9160c8 <unavailable> + 45256
15   libswiftCore.so                    0x0000ffff7f84c7b8 Sequence.forEach(_:) + 384
16   PiTest                             0x0000aaaadd915c68 <unavailable> + 44136
17   libc.so.6                          0x0000ffff7eb74fa8 __libc_start_main + 232
18   PiTest                             0x0000aaaadd9158ec <unavailable> + 43244
[1]    8649 trace trap  sudo swift run
```

uraimo, 2020-06-05:
Thanks for reporting this, are you using a 64bits OS and Swift 5.2.4-v4 on both?

uraimo, 2020-06-05:
It should work now if you pull master/1.1.15 and use .RaspberryPi4 as board type on the Pi4.

KevinVitale, 2020-06-07:
@uraimo thanks for such a quick response. Unfortunately, it's still crashing:

```swift
➜  PiTest sudo swift run
(key: SwiftyGPIO.GPIOName.P12, value: GPIO12<IN,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P22, value: GPIO22<IN,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P6, value: GPIO6<IN,NONE,true>: 1)
(key: SwiftyGPIO.GPIOName.P13, value: GPIO13<IN,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P3, value: GPIO3<OUT,NONE,true>: 1)
(key: SwiftyGPIO.GPIOName.P4, value: GPIO4<IN,NONE,true>: 1)
(key: SwiftyGPIO.GPIOName.P9, value: GPIO9<OUT,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P14, value: GPIO14<OUT,NONE,true>: 1)
(key: SwiftyGPIO.GPIOName.P5, value: GPIO5<IN,NONE,true>: 1)
(key: SwiftyGPIO.GPIOName.P10, value: GPIO10<OUT,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P11, value: GPIO11<OUT,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P20, value: GPIO20<IN,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P15, value: GPIO15<OUT,NONE,true>: 1)
(key: SwiftyGPIO.GPIOName.P17, value: GPIO17<IN,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P26, value: GPIO26<IN,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P24, value: GPIO24<IN,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P19, value: GPIO19<IN,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P23, value: GPIO23<IN,NONE,true>: 0)
(key: SwiftyGPIO.GPIOName.P27, value: GPIO27<IN,NONE,true>: 0)
Fatal error: Unexpectedly found nil while unwrapping an Optional value: file /home/ubuntu/Swift/PiTest/.build/checkouts/SwiftyGPIO/Sources/SwiftyGPIO.swift, line 73
Current stack trace:
0    libswiftCore.so                    0x0000ffff8aa30e4c swift_reportError + 64
1    libswiftCore.so                    0x0000ffff8aa960e4 _swift_stdlib_reportFatalErrorInFile + 128
2    libswiftCore.so                    0x0000ffff8a7ea7b0 <unavailable> + 1439664
3    libswiftCore.so                    0x0000ffff8a7e9a28 _assertionFailure(_:_:file:line:flags:) + 464
4    PiTest                             0x0000aaaac39decd8 <unavailable> + 343256
5    PiTest                             0x0000aaaac39e34b0 <unavailable> + 361648
6    PiTest                             0x0000aaaac39e3628 <unavailable> + 362024
7    libswiftCore.so                    0x0000ffff8a82f5e0 _debugPrint_unlocked<A, B>(_:_:) + 304
8    libswiftCore.so                    0x0000ffff8a899120 <unavailable> + 2154784
9    libswiftCore.so                    0x0000ffff8a82ecb4 _print_unlocked<A, B>(_:_:) + 740
10   libswiftCore.so                    0x0000ffff8a89f5bc <unavailable> + 2180540
11   libswiftCore.so                    0x0000ffff8a89e680 <unavailable> + 2176640
12   PiTest                             0x0000aaaac39960a8 <unavailable> + 45224
13   PiTest                             0x0000aaaac3996148 <unavailable> + 45384
14   PiTest                             0x0000aaaac39961a8 <unavailable> + 45480
15   libswiftCore.so                    0x0000ffff8a8b07b8 Sequence.forEach(_:) + 384
16   PiTest                             0x0000aaaac3995d48 <unavailable> + 44360
17   libc.so.6                          0x0000ffff89bd8fa8 __libc_start_main + 232
18   PiTest                             0x0000aaaac39959cc <unavailable> + 43468
[1]    3010 trace trap  sudo swift run
```
<br/>

This is after I pulled `master` and changed the source to `GPIOs(for: .RaspberryPi4)`:
```
➜  PiTest cat Package.resolved 
{
  "object": {
    "pins": [
      {
        "package": "SwiftyGPIO",
        "repositoryURL": "https://github.com/uraimo/SwiftyGPIO.git",
        "state": {
          "branch": "master",
          "revision": "7c0c84aaae8c035ad0c0bbefe925cb7b91e68ca7",
          "version": null
        }
      }
    ]
  },
  "version": 1
}
```
<hr/>

One other thing: running on **RPi OS 20.04**

KevinVitale, 2020-06-07:
**Update:** Appears it may be an _Ubuntu Server 20.04_ issue? I had a suspicion, and when I re-ran my script on RPI 4B w/Ubuntu Server 18.04, it worked.

My apologies; I should restate my test rigs (for us both):

- RPi 3B+, 1GB – Ubuntu Server 18.04 ✅
- RPi 4B, 4GB – Ubuntu Server 18.04 ✅
- RPi 4B, 4GB – Ubuntu Server 20.04 ❌
- RPi 4B, 8GB – Ubuntu Server 20.04 ❌

<br/>

uraimo, 2020-06-07:
I'm getting the same error on a Pi4B Rev1.1 with 32bit debian buster (similar to 18.04), the issue seem to be consistently with the initialization of gpio 10,11 and 19.
Updating the firmware/OS didn't help, I need to investigate this further.

uraimo, 2020-06-07:
Those are all SPI/PWM pins, can you install wiringpi and run "gpio readall" on 18.04 and 20.04?
Since I can't even export those gpios manually, it could be that they are locked in the alternative mode now, readall should print the current mode for each one.

KevinVitale, 2020-06-08:
```
Ubuntu 20.04 LTS (GNU/Linux 5.4.0-1011-raspi aarch64)

➜  WiringPi git:(master) sudo gpio readall  
 +-----+-----+---------+------+---+---Pi 4B--+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 | ALT0 | 1 |  3 || 4  |   |      | 5v      |     |     |
 |   3 |   9 |   SCL.1 | ALT0 | 1 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   IN | 1 |  7 || 8  | 1 | ALT5 | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 1 | ALT5 | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 |   IN | 0 | 11 || 12 | 0 | IN   | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 |   IN | 0 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 |   IN | 0 | 15 || 16 | 0 | IN   | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 0 | IN   | GPIO. 5 | 5   | 24  |
 |  10 |  12 |    MOSI | ALT0 | 0 | 19 || 20 |   |      | 0v      |     |     |
 |   9 |  13 |    MISO | ALT0 | 0 | 21 || 22 | 0 | IN   | GPIO. 6 | 6   | 25  |
 |  11 |  14 |    SCLK | ALT0 | 0 | 23 || 24 | 1 | OUT  | CE0     | 10  | 8   |
 |     |     |      0v |      |   | 25 || 26 | 1 | OUT  | CE1     | 11  | 7   |
 |   0 |  30 |   SDA.0 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.0   | 31  | 1   |
 |   5 |  21 | GPIO.21 |   IN | 1 | 29 || 30 |   |      | 0v      |     |     |
 |   6 |  22 | GPIO.22 |   IN | 1 | 31 || 32 | 0 | IN   | GPIO.26 | 26  | 12  |
 |  13 |  23 | GPIO.23 |   IN | 0 | 33 || 34 |   |      | 0v      |     |     |
 |  19 |  24 | GPIO.24 |   IN | 0 | 35 || 36 | 0 | IN   | GPIO.27 | 27  | 16  |
 |  26 |  25 | GPIO.25 |   IN | 0 | 37 || 38 | 0 | IN   | GPIO.28 | 28  | 20  |
 |     |     |      0v |      |   | 39 || 40 | 0 | IN   | GPIO.29 | 29  | 21  |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+---Pi 4B--+---+------+---------+-----+-----+
➜  WiringPi git:(master) gpio -v          
gpio version: 2.60
Copyright (c) 2012-2018 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Pi 4B, Revision: 04, Memory: 0MB, Maker: Sony 
  * Device tree is enabled.
  *--> Raspberry Pi 4 Model B Rev 1.4
  * This Raspberry Pi supports user-level GPIO access.
```
<br/>

```
Ubuntu 18.04.4 LTS (GNU/Linux 5.3.0-1026-raspi2 aarch64)

➜  WiringPi git:(master) sudo gpio readall
 +-----+-----+---------+------+---+---Pi 4B--+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 | ALT0 | 1 |  3 || 4  |   |      | 5v      |     |     |
 |   3 |   9 |   SCL.1 | ALT0 | 1 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   IN | 1 |  7 || 8  | 1 | ALT5 | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 1 | ALT5 | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 |   IN | 0 | 11 || 12 | 0 | IN   | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 |   IN | 0 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 |   IN | 0 | 15 || 16 | 0 | IN   | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 0 | IN   | GPIO. 5 | 5   | 24  |
 |  10 |  12 |    MOSI | ALT0 | 0 | 19 || 20 |   |      | 0v      |     |     |
 |   9 |  13 |    MISO | ALT0 | 0 | 21 || 22 | 0 | IN   | GPIO. 6 | 6   | 25  |
 |  11 |  14 |    SCLK | ALT0 | 0 | 23 || 24 | 1 | OUT  | CE0     | 10  | 8   |
 |     |     |      0v |      |   | 25 || 26 | 1 | OUT  | CE1     | 11  | 7   |
 |   0 |  30 |   SDA.0 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.0   | 31  | 1   |
 |   5 |  21 | GPIO.21 |   IN | 1 | 29 || 30 |   |      | 0v      |     |     |
 |   6 |  22 | GPIO.22 |   IN | 1 | 31 || 32 | 0 | IN   | GPIO.26 | 26  | 12  |
 |  13 |  23 | GPIO.23 |   IN | 0 | 33 || 34 |   |      | 0v      |     |     |
 |  19 |  24 | GPIO.24 |   IN | 0 | 35 || 36 | 0 | IN   | GPIO.27 | 27  | 16  |
 |  26 |  25 | GPIO.25 |   IN | 0 | 37 || 38 | 0 | IN   | GPIO.28 | 28  | 20  |
 |     |     |      0v |      |   | 39 || 40 | 0 | IN   | GPIO.29 | 29  | 21  |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+---Pi 4B--+---+------+---------+-----+-----+
➜  WiringPi git:(master) gpio -v
gpio version: 2.60
Copyright (c) 2012-2018 Gordon Henderson
This is free software with ABSOLUTELY NO WARRANTY.
For details type: gpio -warranty

Raspberry Pi Details:
  Type: Pi 4B, Revision: 02, Memory: 4096MB, Maker: Sony 
  * Device tree is enabled.
  *--> Raspberry Pi 4 Model B Rev 1.2
  * This Raspberry Pi supports user-level GPIO access.
```

KevinVitale, 2020-06-21:
@uraimo anything I can do to help with this?

uraimo, 2020-08-02:
Excluded the gpio 10, 11 and 19 from the Pi4's list until Raspian fixes them upstream.
The issue can be replicated via terminal too, exporting one of the three gpios.