uraimo/SwiftyGPIO #106: GPIO19 missing for RaspberryPi4

simon2204, 2020-08-03:
### Board Type

RaspberryPi4

### Operating System

Raspbian

### Swift Version

Swift version 5.1.5 (swift-5.1.5-RELEASE)

### Description

GPIO19 for RaspberryPi4 is not included in GPIORPI4

simon2204, 2020-08-03:
I just saw that you excluded the gpio 10, 11 and 19 on purpose

closed 2020-08-03 by simon2204

uraimo, 2020-08-03:
Hi @simon2204 it's for #104, on the Pi4 gpios 10,11 and 19 cannot be initialized correctly for an issue with the gpio sysfs driver or even the firmware, so I've excluded them from the list for the Pi4.
I haven't investigated this further and so I can't say if they fixed it or if anyone is even looking into this from their side (issue replicable manually doing something like echo "19" > /sys/class/gpio/export).
If you want to try if something changed, you can initialize the gpio manually with "RaspberryGPIO(name:"GPIO19", id:19, baseAddr:0x7E000000)", the line that was removed from the presets.

simon2204, 2020-08-04:
@uraimo Yes thank you, I already did and it works fine for me.

uraimo, 2020-08-04:
@simon2204 oh, you mean that gpio 19 is working correctly when initialized that way? last time I checked was 2 months ago.

simon2204, 2020-08-04:
@uraimo I just started learning to work with the Pi and I have no problem with GPIO19