uraimo/SwiftyGPIO #107: SPI fails on 32-bit Zero

barretpj, 2020-08-12:
### Board Type

Raspberry Pi Zero W

### Operating System

Raspberry Pi OS Lite
Linux raspberrypi 5.4.51+ #1327 Thu Jul 23 10:53:06 BST 2020 armv6l GNU/Linux

### Swift Version

Pre-built from https://github.com/uraimo/buildSwiftOnARM/releases/download/5.1.5/swift-5.1.5-armv6-RPi01234-RaspbianBuster.tgz

### Description

SPI calls fail with "Message too long" regardless of the length of message.

On investigation it's because `struct spi_ioc_transfer` uses `UnsafeMutableRawPointer` for `tx_buf` and `rx_buf`, but `/usr/include/linux/spi/spidev.h` defines these pointers as `__u64`

If I add 32 bits of padding after these 2 members in to `struct spi_ioc_transfer` the SPI calls succeed.

gestrich, 2020-08-16:
From testing various versions, this bugs seem introduced in v 1.1.15. Version. 1.1.14 seems ok.

uraimo, 2020-09-01:
Thanks @barretpj @gestrich, exactly, 1.1.15 introduced [this](https://github.com/uraimo/SwiftyGPIO/commit/9066d5dfc58fb53c325b07762e38424c3a9de230#diff-660c0bd869b9f2598f5daaf80617f88fL89-L90) change that broke it and the fix is converting to an uint64 via bitpattern (that 
 actually adds a padding at the beginning if necessary), will fix/test it as soon as I can get a hold of a pi.