uraimo/SwiftyGPIO #109: Unable to send commands to Nextion display via UART

lairdtrimble, 2020-09-12:
### Board Type

RaspberryPi4

### Operating System

Raspbian

### Swift Version

5.1.5 prebuilt

### Description

I have connected a Nextion Display to a Pi via the UART interface. I can successfully read data from the device. However I am unsuccessful at sending commands to the device. Commands are short text strings terminated with ÿÿÿ (0xFF, 0xFF, 0xFF). For instance, the string "page 2ÿÿÿ" switches the display to page 2. When I look at the data that is sent to the UART for this command I see:

[112, 97, 103, 101, 32, 50, -61, -65, -61, -65, -61, -65, 0]

It looks like the "ÿ" terminator characters are being converted into two bytes (-61, -65) instead of 0xFF. I presume that this is because the individual characters are being converted to the CChar (Int8) data type and something is happening with the sign bit. Also, the C end of string value of 0 is appended.

How do I go about sending a string terminated with "ÿÿÿ"?

I am a relatively new Swift programmer and very new at working with the Pi and SwiftyGPIO so please forgive me if this is a newbie question with an obvious answer.

Laird

lairdtrimble, 2020-09-14:
Followup:

Sorry to bother everybody with this newbie question. With a little reading and a bit of experimentation I resolved this issue myself.

Laird

closed 2020-09-14 by lairdtrimble

uraimo, 2020-09-14:
Hi @lairdtrimble,  thanks for the follow-up, did you implement a workaround for some issue in the library or in end the write method worked as expected?

lairdtrimble, 2020-09-14:
Hi @uraimo, the library is fine. I made some incorrect assumptions about how Swift strings worked. Once that was sorted out, I was able to write a method that worked as expected. Hopefully when this is done I can upload libraries for the Nextion display device and ADS1115 adc.

uraimo, 2020-09-14:
Great! Feel free to open other issues if you notice something not working.