uraimo/SwiftyGPIO #118: Reading from serial port hangs once I try to interact with GPIO pin

MihaelIsaev, 2021-03-07:
### Board Type

RaspberryPi4

### Operating System

Ubuntu 20.04

### Swift Version

Swift 5.3.3 precompiled from swift-arm.com

### Description

I use uhf-rfid R2000 reader connected via `/dev/ttyUSB0` and I'm continuously reading tags from it.
To connect to R2000 and read tags I use [SwiftSerial](https://github.com/yeokm1/SwiftSerial) lib.

The problem is that once I try to interact with GPIO pin

either
```swift
SwiftyGPIO.GPIOs(for:.RaspberryPi4)[.P22]!.value = 1
```
or
```swift
SwiftyGPIO.GPIOs(for:.RaspberryPi4)[.P22]!.value = 0
```
I see that SwiftSerial hangs on [POSIX read command](https://github.com/yeokm1/SwiftSerial/blob/4e46944053c9dd5b0340aeed4772d68bc371c04e/Sources/SwiftSerial.swift#L385).

If I don't interact with GPIO then everything works without any issues.

I tried to call all the commands inside one `DispatchQueue` with `.sync { ... }` block, to be sure that they're calls one by one synchronously (like not in parallel), but it doesn't help cause it still hangs.

I also created [an issue in SwiftSerial repo](https://github.com/yeokm1/SwiftSerial/issues/20)

Trying to figure out what's wrong, any help would be appreciated!

uraimo, 2021-03-07:
Hi, just to be 100% sure and ignore this if you already checked, but can you double check that when you enabled the uart you also disabled the login+kernel messages on the uart port in raspi-config?

MihaelIsaev, 2021-04-06:
@uraimo thank you for the response and sorry for the long silence. I tried a lot of things and `login+kernel` on uart is disabled for sure. External device works pretty well, so the problem is only with hanging gpio on raspberry. A week ago I switched from RPi to ESP32 and here everything works well without and hanging. So I'm giving up with raspberry and its glitches...

closed 2021-04-06 by MihaelIsaev