uraimo/SwiftyGPIO #126: UART runtime error

ea7kir, 2021-11-22:
### Board Type: RaspberryPi4
### Operating System: raspios_lite_arm64
### Swift Version: 5.5
### Description

Just a new Package with Hello world!.
   import Foundation
   // NB import Glibc doesn't make any difference on this project or others I've built.
   import SwiftyGPIO
   print("Hello, world!")

RUN TIME ERRORS:

/home/pi/IOTests/.build/checkouts/SwiftyGPIO/Sources/UART.swift:44:31: error: consecutive statements on a line must be separated by ';'
             .RaspberryPiZero2,
                              ^
                              ;
/home/pi/IOTests/.build/checkouts/SwiftyGPIO/Sources/UART.swift:44:31: error: expected expression
             .RaspberryPiZero2,
                              ^
/home/pi/IOTests/.build/checkouts/SwiftyGPIO/Sources/UART.swift:44:15: error: reference to member 'RaspberryPiZero2' cannot be resolved without a contextual type
             .RaspberryPiZero2,
             ~^~~~~~~~~~~~~~~~
[10/13] Compiling SwiftyGPIO Presets.swift
error: fatalError

STRANGE:

Another project, which is much bigger, compiles and runs without errors.

uraimo, 2021-11-22:
Hi, forgot to push that 1.3.7 tag, is fixed now, thanks for reporting it.

closed 2021-11-22 by uraimo

ea7kir, 2021-11-22:
Hi,

Have you pushed 1.3.7 yet, because I’m still getting the same error?

UART.swift was by curuvar Fixed a couple of bugs in 1.3.6 (#124) on the main branch.

Sorry, but I’m a novice with Github.

Regards,
Michael Naylor - EA7KIR
https://michaelnaylor.es

uraimo, 2021-11-22:
Hi, yes, you need to do a clean build to pull the new release.

ea7kir, 2021-11-22:
Yes I’ve been doing swift package clean and swift package purge-cache.
Package has:  .package(url: "https://github.com/uraimo/SwiftyGPIO.git", from: "1.0.0”),

What I don’t understand is that I have a larger project that doesn’t produce any errors, even when I clean and purge.

Regards,
Michael Naylor

ea7kir, 2021-11-22:
Does this mean anything to you?

Regards,
Michael Naylor

ea7kir, 2021-11-22:
> Does this mean anything to you?
>
> <Screenshot 2021-11-22 at 13.46.39.png>
>
> Regards,
> Michael Naylor

Maybe I should mention I’m on macOS 12.0.1 Monterey with Xcode 13.1

Regards,
Michael Naylor

uraimo, 2021-11-22:
I'd try removing completely the .build directory before trying again (the library is being cloned to .build/checkouts/SwiftyGPIO/ that is a full git clone), that error was fixed in the commit you pasted above. I'll double check as soon as I get access to a machine with swift in a few hours.

ea7kir, 2021-11-22:
***@***.***:~/IOTests $ rm -rf .build
***@***.***:~/IOTests $ ls -la
total 52
drwxr-xr-x  5 pi pi 4096 Nov 22 14:03 .
drwxr-xr-x  7 pi pi 4096 Nov 22 10:21 ..
-rw-r--r--  1 pi pi  126 Nov 22 10:21 .gitignore
-rw-r--r--+ 1 pi pi  326 Nov 22 13:59 Package.resolved
-rw-r--r--+ 1 pi pi  885 Nov 22 11:47 Package.swift
-rw-r--r--  1 pi pi   42 Nov 22 10:21 README.md
drwxr-xr-x  3 pi pi 4096 Nov 22 10:21 Sources
drwxr-xr-x+ 3 pi pi 4096 Nov 22 10:23 .swiftpm
drwxr-xr-x  3 pi pi 4096 Nov 22 10:21 Tests
***@***.***:~/IOTests $ swift build
Fetching https://github.com/uraimo/SwiftyGPIO.git from cache
Fetched https://github.com/uraimo/SwiftyGPIO.git (0.67s)
Computing version for https://github.com/uraimo/SwiftyGPIO.git
Computed https://github.com/uraimo/SwiftyGPIO.git at 1.3.6 (0.03s)
Creating working copy for https://github.com/uraimo/SwiftyGPIO.git
Working copy of https://github.com/uraimo/SwiftyGPIO.git resolved at 1.3.6
/home/pi/IOTests/.build/checkouts/SwiftyGPIO/Sources/UART.swift:44:31: error: consecutive statements on a line must be separated by ';'
             .RaspberryPiZero2,
                              ^
                              ;
/home/pi/IOTests/.build/checkouts/SwiftyGPIO/Sources/UART.swift:44:31: error: expected expression
             .RaspberryPiZero2,
                              ^
/home/pi/IOTests/.build/checkouts/SwiftyGPIO/Sources/UART.swift:44:15: error: reference to member 'RaspberryPiZero2' cannot be resolved without a contextual type
             .RaspberryPiZero2,
             ~^~~~~~~~~~~~~~~~
/home/pi/IOTests/.build/checkouts/SwiftyGPIO/Sources/UART.swift:44:31: error: consecutive statements on a line must be separated by ';'
             .RaspberryPiZero2,
                              ^
                              ;
/home/pi/IOTests/.build/checkouts/SwiftyGPIO/Sources/UART.swift:44:31: error: expected expression
             .RaspberryPiZero2,
                              ^
/home/pi/IOTests/.build/checkouts/SwiftyGPIO/Sources/UART.swift:44:15: error: reference to member 'RaspberryPiZero2' cannot be resolved without a contextual type
             .RaspberryPiZero2,
             ~^~~~~~~~~~~~~~~~
[10/13] Compiling SwiftyGPIO I2C.swift
***@***.***:~/IOTests $ 

Regards,
Michael Naylor

uraimo, 2021-11-22:
Ok, I guess the cache it's now under .swiftpm, sorry but I haven't touched this in a while.

uraimo, 2021-11-22:
(It's getting it from cache, and still resolving it as 1.3.6)

ea7kir, 2021-11-22:
But who’s cache?  I’ve reload Safari cache several times, but the GitHub always talks about 1.3.6

Don’t think .swiftpm has anything to do with this.  It’s there because I edit the source in Xcode on the Mac, but I compile on the RPi.

Maybe it’s a bug on GitHub.com

Regards,
Michael Naylor

uraimo, 2021-11-22:
https://github.com/uraimo/SwiftyGPIO/tags the last one is 1.3.7.

From your local run:

```
***@***.***:~/IOTests $ swift build
Fetching https://github.com/uraimo/SwiftyGPIO.git from cache <-------------
Fetched https://github.com/uraimo/SwiftyGPIO.git (0.67s)
Computing version for https://github.com/uraimo/SwiftyGPIO.git
Computed https://github.com/uraimo/SwiftyGPIO.git at 1.3.6 (0.03s)
```

uraimo, 2021-11-22:
Pretty sure it's the swiftpm cache, that they likely moved from .build to .swiftpm.

ea7kir, 2021-11-22:
Same result…

***@***.***:~/IOTests $ rm -rf .build
***@***.***:~/IOTests $ rm -rf .swiftpm
***@***.***:~/IOTests $ ls -la
total 44
drwxr-xr-x  4 pi pi 4096 Nov 22 14:56 .
drwxr-xr-x  7 pi pi 4096 Nov 22 10:21 ..
-rw-r--r--  1 pi pi  126 Nov 22 10:21 .gitignore
-rw-r--r--+ 1 pi pi  326 Nov 22 14:03 Package.resolved
-rw-r--r--+ 1 pi pi  885 Nov 22 11:47 Package.swift
-rw-r--r--  1 pi pi   42 Nov 22 10:21 README.md
drwxr-xr-x  3 pi pi 4096 Nov 22 10:21 Sources
drwxr-xr-x  3 pi pi 4096 Nov 22 10:21 Tests
***@***.***:~/IOTests $ swift build
Fetching https://github.com/uraimo/SwiftyGPIO.git from cache
Fetched https://github.com/uraimo/SwiftyGPIO.git (0.86s)
Computing version for https://github.com/uraimo/SwiftyGPIO.git
Computed https://github.com/uraimo/SwiftyGPIO.git at 1.3.6 (0.47s)
Creating working copy for https://github.com/uraimo/SwiftyGPIO.git
Working copy of https://github.com/uraimo/SwiftyGPIO.git resolved at 1.3.6
/home/pi/IOTests/.build/checkouts/SwiftyGPIO/Sources/UART.swift:44:31: error: consecutive statements on a line must be separated by ';'
             .RaspberryPiZero2,
                              ^
                              ;
/home/pi/IOTests/.build/checkouts/SwiftyGPIO/Sources/UART.swift:44:31: error: expected expression
             .RaspberryPiZero2,
                              ^
/home/pi/IOTests/.build/checkouts/SwiftyGPIO/Sources/UART.swift:44:15: error: reference to member 'RaspberryPiZero2' cannot be resolved without a contextual type
             .RaspberryPiZero2,
             ~^~~~~~~~~~~~~~~~
/home/pi/IOTests/.build/checkouts/SwiftyGPIO/Sources/UART.swift:44:31: error: consecutive statements on a line must be separated by ';'
             .RaspberryPiZero2,
                              ^
                              ;
/home/pi/IOTests/.build/checkouts/SwiftyGPIO/Sources/UART.swift:44:31: error: expected expression
             .RaspberryPiZero2,
                              ^
/home/pi/IOTests/.build/checkouts/SwiftyGPIO/Sources/UART.swift:44:15: error: reference to member 'RaspberryPiZero2' cannot be resolved without a contextual type
             .RaspberryPiZero2,
             ~^~~~~~~~~~~~~~~~
[10/13] Compiling SwiftyGPIO I2C.swift
***@***.***:~/IOTests $ 

Regards,
Michael Naylor

curuvar, 2021-11-22:
To force a complete build remove the .build directory and the Package.resolved file.   If you don’t remove the latter, swift wants to re-download the original version.

The may be a more proper way of doing this, but this works for me.

ea7kir, 2021-11-22:
Problem solved, except we’re now getting 1.3.8 !

***@***.***:~/IOTests $ rm -rf .build
***@***.***:~/IOTests $ rm Package.resolved 
***@***.***:~/IOTests $ swift build
Fetching https://github.com/uraimo/SwiftyGPIO.git from cache
Fetched https://github.com/uraimo/SwiftyGPIO.git (0.87s)
Computing version for https://github.com/uraimo/SwiftyGPIO.git
Computed https://github.com/uraimo/SwiftyGPIO.git at 1.3.8 (0.02s)
Creating working copy for https://github.com/uraimo/SwiftyGPIO.git
Working copy of https://github.com/uraimo/SwiftyGPIO.git resolved at 1.3.8
[21/21] Build complete!
***@***.***:~/IOTests $ swift run
[0/0] Build complete!
Hello, world!
***@***.***:~/IOTests $ 

Regards,
Michael Naylor

uraimo, 2021-11-22:
That's correct, just pushed it :)