uraimo/SwiftyGPIO #128: SPI fails on RPI 3 model B

gacobj, 2021-12-03:
### Board Type

RaspberryPi3

### Operating System

Raspbian Buster

### Swift Version

swift-5.1.5

### Description

SPI send call fails with "Message too long" error.

### This appears to fix this issue.

struct spi_ioc_transfer {
        //var tx_buf: UnsafeMutableRawPointer
        //var rx_buf: UnsafeMutableRawPointer
        var tx_buf: UInt64
        var rx_buf: UInt64
        var len: UInt32
        var speed_hz: UInt32 = 500000
        var delay_usecs: UInt16 = 0
        var bits_per_word: UInt8 = 8
        let cs_change: UInt8 = 0
        let tx_nbits: UInt8 = 0
        let rx_nbits: UInt8 = 0
        let pad: UInt16 = 0
    }

And in transferData function

var r: CInt = 0
        
        r = tx.withUnsafeMutableBufferPointer { (txPtr: inout UnsafeMutableBufferPointer<UInt8>) -> CInt in
            if let txPtr1 = UnsafeMutableRawPointer(txPtr.baseAddress) {
                return rx.withUnsafeMutableBufferPointer { (rxPtr : inout UnsafeMutableBufferPointer<UInt8>) -> CInt in
                    if let rxPtr1 = UnsafeMutableRawPointer(rxPtr.baseAddress) {
                        var tr = spi_ioc_transfer(
                            //tx_buf: txPtr.baseAddress!,
                            tx_buf: UInt64(UInt(bitPattern: txPtr1)),
                            //rx_buf: rxPtr.baseAddress!,
                            rx_buf: UInt64(UInt(bitPattern: rxPtr1)),
                            len: UInt32(txPtr.count),
                            speed_hz: speed,
                            delay_usecs: delay,
                            bits_per_word: bits)
                        return ioctl(fd, SPI_IOC_MESSAGE1, &tr)
                    }
                    return 0
                }
            }
            return 0
        }

uraimo, 2021-12-07:
Thanks! I'll do some tests before integrating the fix, can you confirm that you are using a 32bit debian?

uraimo, 2021-12-08:
Ok, this was seen before with #107, spidev.h changed last year and that structure need to be migrated to u64, will do.

gacobj, 2021-12-09:
I am not sure if it is a 32-bit debian. This is what I'm running:

Linux version 5.10.63-v7+ (dom@buildbot) (arm-linux-gnueabihf-gcc-8 (Ubuntu/Linaro 8.4.0-3ubuntu1) 8.4.0, GNU ld (GNU Binutils for Ubuntu) 2.34) #1488 SMP Thu Nov 18 16:14:44 GMT 2021