uraimo/SwiftyGPIO #133: 64bit PWM broken

apocolipse, 2022-05-30:
### Board Type
RaspberryPi 4 4GB

### Operating System
Raspian Bullseye (64bit)

### Swift Version

5.6.1 Release (Official, aarch64-unknown-linux-gnu)

### Description

Using PWM results in a fatalerror, `Fatal error: ioctl on mailbox failed!`.  This happens for all PWMs
Looking into the code, it appears `IOCTL_MBOX_PROPERTY` is wrong.  Its currently hard coded to `0xc0046400`
```swift
 private let IOCTL_MBOX_PROPERTY: UInt = 0xc0046400 // _IOWR(100, 0, char *)
 ```
From the C version they call `_IOWR(100, 0, char *)` which you have commented out, which results in  avalue of `0xc0086400` since `char *` size is different. 

Changing the hard coded value works**, it gets past the fatal error just fine, Heres some code to have it be generic regardless of OS
```swift
// ioctl helpers?
fileprivate enum IOCTL {
  static let IOCPARM_MASK: UInt16 =  0x1fff
  static let IOC_VOID: UInt32 = 0x20000000
  static let IOC_OUT: UInt32 = 0x40000000
  static let IOC_IN: UInt32 = 0x80000000
  static let IOC_INOUT: UInt32 = IOC_IN | IOC_OUT
  static let IOC_DIRMASK: UInt32 = 0xe0000000

  static func _IOC (_ io: UInt32, _ group: UInt32, _ num: UInt32, _ len: UInt32) -> UInt32 {
    let rv = io | (( len & UInt32(IOCPARM_MASK)) << 16) | ((group << 8) | num)
    return rv
  }
  static func _IOWR (_ group: UInt32, _ num: UInt32, _ size: UInt32) -> UInt32 {
    return _IOC(IOC_INOUT, group, num, size)
  }
}
...
    private let IOCTL_MBOX_PROPERTY: UInt =  UInt(IOCTL._IOWR(100, 0, UInt32(MemoryLayout<Int>.size)))
```

However, with this now fixed, it's now crashing with `bus error`, specifically crashing around [PWM.swift#414](https://github.com/uraimo/SwiftyGPIO/blob/master/Sources/PWM.swift#L414) where its trying to copy data.  
This is where I've hit a roadblock.  Not sure what to do to keep debugging.  
Currently, i'm looking to use the rpi-ws281x c library directly in Swift as an alternative for my current project, as I've tested and it works just fine with Python bindings.  Would love to have a pure swift soln working in 64bit OS's though

AndrewConnick, 2023-04-30:
I have the same problem on a RaspberryPi Zero 2 W running Swift 5.8 on Ubuntu 23.04. Fixing the IOCTL_MBOX_PROPERTY, as above, I also get "Bus error" when copying memory.

It has been a long time since the last update to SwiftyGPIO. Is it still being monitored/updated? Has it been superseded by another package? I would love to get my NeoPixels working!

kbongort, 2023-11-26:
I had the same issue, and ditched SwiftyGPIO for a thin wrapper around the [rpi_ws281x](https://github.com/jgarff/rpi_ws281x) library. My wrapper is published as a Swift package at https://github.com/kbongort/rpi-ws281x-swift.

This has the benefit of building the C code from source, so it always uses current system header values.