uraimo/SwiftyGPIO #137: Package.swift version is incompatible with 32 bit compilers

nallick, 2024-03-11:
### Operating System

Raspian Buster, 32-bit

### Swift Version

5.1.5

### Description

The Package.swift file has been updated to swift-tools-version 5.4 despite not including any new features that require that version. I'm trying to support a RPi2, which is 32-bit only hardware. Therefore the most recent version of Swift for it is v5.1.5. When I try to update my package I get the compiler error:

error: the package dependency graph could not be resolved; unable to find any available tag

Reverting the SwiftyGPIO package tools version to 5.1 solves this problem and continues to be backward compatible with newer versions of Swift since there's nothing in the package definition that needs the newer tools version.

uraimo, 2024-03-12:
Hi, I've released a new version, where did you get that 5.1.5 build?

closed 2024-03-12 by uraimo

nallick, 2024-03-12:
Thanks for taking care of that quickly.

5.1.5 seems to be the final supported 32-bit build at [Swift-Arm](https://swift-arm.com/installSwift). There are details [here](https://forums.swift.org/t/new-swift-package-repository-for-ubuntu-debian-linux-distributions/50412).

> Available from the Swift-Arm package repository
Debian 10 Buster
Debian 11 Bullseye
Ubuntu 18.04 / 20.04 / 22.04

> Raspberry Pi OS / Raspbian Buster (32 bit)
> Raspberry Pi OS / Raspbian Bullseye (64 bit)

commit a54de497 (2024-03-12): Lower minimum supported version to 5.1, closes #137
commit 73883a22 (2024-03-15): Revert "Lower minimum supported version to 5.1, closes #137"

This reverts commit a54de4975eee388fd5c594b01c9cd15dbe96ddea.
commit a41e7a42 (2024-03-15): Again, try to set 5.1 as minimum version, closes #137