uraimo/SwiftyGPIO #140: Compatibility: raspberry pi 5? swift 5.10

fishandphil, 2024-08-14:
We're excited about using SwiftyGPIO in our new project which involves a Raspberry Pi. We currently have the following configuration:

- Raspberry Pi 5 
- Swift-tools-version 5.10
- Ubuntu 24.04 (noble)

We're hitting the following errors when adding the SwiftyGPIO package. I am wondering if it is because of Pi 5? Swift 5.1 or Ubuntu 24.04? On the main page, Pi 5 is not listed as compatible. 
I can read that Swift 5.x is supported. 

Here is our package.swift file:

`// swift-tools-version: 5.10
// The swift-tools-version declares the minimum version of Swift required to build this package.

import PackageDescription

let package = Package(
    name: "PIRSensorProject",
    dependencies: [
        .package(url: "https://github.com/uraimo/SwiftyGPIO.git", from: "1.0.0")
    ],
    targets: [
        // Targets are the basic building blocks of a package, defining a module or a test suite.
        // Targets can depend on other targets in this package and products from dependencies.
        .executableTarget(
            name: "PIRSensorProject",
            dependencies: ["SwiftyGPIO"]),
    ]
)`

The package update command run smoothly but once we build our empty main.swift file, we get the error:

`philippe@oursler:~/Desktop/PIRSensorProject$ swift build
Building for debugging...
/home/philippe/Desktop/PIRSensorProject/.build/checkouts/SwiftyGPIO/Sources/SwiftyGPIO.swift:175:24: error: value of optional type 'UnsafeMutablePointer<FILE>?' (aka 'Optional<UnsafeMutablePointer<_IO_FILE>>') must be unwrapped to a value of type 'UnsafeMutablePointer<FILE>' (aka 'UnsafeMutablePointer<_IO_FILE>')
        defer { fclose(fp) }
                       ^
/home/philippe/Desktop/PIRSensorProject/.build/checkouts/SwiftyGPIO/Sources/SwiftyGPIO.swift:175:24: note: coalesce using '??' to provide a default when the optional value contains 'nil'
        defer { fclose(fp) }
                       ^
                          ?? <#default value#>
/home/philippe/Desktop/PIRSensorProject/.build/checkouts/SwiftyGPIO/Sources/SwiftyGPIO.swift:175:24: note: force-unwrap using '!' to abort execution if the optional value contains 'nil'
        defer { fclose(fp) }
                       ^
                         !
/home/philippe/Desktop/PIRSensorProject/.build/checkouts/SwiftyGPIO/Sources/SwiftyGPIO.swift:178:64: error: value of optional type 'UnsafeMutablePointer<FILE>?' (aka 'Optional<UnsafeMutablePointer<_IO_FILE>>') must be unwrapped to a value of type 'UnsafeMutablePointer<FILE>' (aka 'UnsafeMutablePointer<_IO_FILE>')
            return fwrite(buffer.baseAddress, buffer.count, 1, fp) - buffer.count
                                                               ^
/home/philippe/Desktop/PIRSensorProject/.build/checkouts/SwiftyGPIO/Sources/SwiftyGPIO.swift:178:64: note: coalesce using '??' to provide a default when the optional value contains 'nil'
            return fwrite(buffer.baseAddress, buffer.count, 1, fp) - buffer.count
                                                               ^
                                                                  ?? <#default value#>
/home/philippe/Desktop/PIRSensorProject/.build/checkouts/SwiftyGPIO/Sources/SwiftyGPIO.swift:178:64: note: force-unwrap using '!' to abort execution if the optional value contains 'nil'
            return fwrite(buffer.baseAddress, buffer.count, 1, fp) - buffer.count
                                                               ^
                                                                 !
/home/philippe/Desktop/PIRSensorProject/.build/checkouts/SwiftyGPIO/Sources/SwiftyGPIO.swift:181:23: error: value of optional type 'UnsafeMutablePointer<FILE>?' (aka 'Optional<UnsafeMutablePointer<_IO_FILE>>') must be unwrapped to a value of type 'UnsafeMutablePointer<FILE>' (aka 'UnsafeMutablePointer<_IO_FILE>')
            if ferror(fp) != 0 {
                      ^
/home/philippe/Desktop/PIRSensorProject/.build/checkouts/SwiftyGPIO/Sources/SwiftyGPIO.swift:181:23: note: coalesce using '??' to provide a default when the optional value contains 'nil'
            if ferror(fp) != 0 {
                      ^
                         ?? <#default value#>
/home/philippe/Desktop/PIRSensorProject/.build/checkouts/SwiftyGPIO/Sources/SwiftyGPIO.swift:181:23: note: force-unwrap using '!' to abort execution if the optional value contains 'nil'
            if ferror(fp) != 0 {
                      ^
                        !
/home/philippe/Desktop/PIRSensorProject/.build/checkouts/SwiftyGPIO/Sources/SwiftyGPIO.swift:193:24: error: value of optional type 'UnsafeMutablePointer<FILE>?' (aka 'Optional<UnsafeMutablePointer<_IO_FILE>>') must be unwrapped to a value of type 'UnsafeMutablePointer<FILE>' (aka 'UnsafeMutablePointer<_IO_FILE>')
        defer { fclose(fp) }
                       ^
/home/philippe/Desktop/PIRSensorProject/.build/checkouts/SwiftyGPIO/Sources/SwiftyGPIO.swift:193:24: note: coalesce using '??' to provide a default when the optional value contains 'nil'
        defer { fclose(fp) }
                       ^
                          ?? <#default value#>
/home/philippe/Desktop/PIRSensorProject/.build/checkouts/SwiftyGPIO/Sources/SwiftyGPIO.swift:193:24: note: force-unwrap using '!' to abort execution if the optional value contains 'nil'
        defer { fclose(fp) }
                       ^
                         !
/home/philippe/Desktop/PIRSensorProject/.build/checkouts/SwiftyGPIO/Sources/SwiftyGPIO.swift:198:60: error: value of optional type 'UnsafeMutablePointer<FILE>?' (aka 'Optional<UnsafeMutablePointer<_IO_FILE>>') must be unwrapped to a value of type 'UnsafeMutablePointer<FILE>' (aka 'UnsafeMutablePointer<_IO_FILE>')
            let len = fread(buffer.baseAddress, MAXLEN, 1, fp)
                                                           ^
/home/philippe/Desktop/PIRSensorProject/.build/checkouts/SwiftyGPIO/Sources/SwiftyGPIO.swift:198:60: note: coalesce using '??' to provide a default when the optional value contains 'nil'
            let len = fread(buffer.baseAddress, MAXLEN, 1, fp)
                                                           ^
                                                              ?? <#default value#>
/home/philippe/Desktop/PIRSensorProject/.build/checkouts/SwiftyGPIO/Sources/SwiftyGPIO.swift:198:60: note: force-unwrap using '!' to abort execution if the optional value contains 'nil'
            let len = fread(buffer.baseAddress, MAXLEN, 1, fp)
                                                           ^
                                                             !
/home/philippe/Desktop/PIRSensorProject/.build/checkouts/SwiftyGPIO/Sources/SwiftyGPIO.swift:198:60: error: value of optional type 'UnsafeMutablePointer<FILE>?' (aka 'Optional<UnsafeMutablePointer<_IO_FILE>>') must be unwrapped to a value of type 'UnsafeMutablePointer<FILE>' (aka 'UnsafeMutablePointer<_IO_FILE>')
            let len = fread(buffer.baseAddress, MAXLEN, 1, fp)
                                                           ^
/home/philippe/Desktop/PIRSensorProject/.build/checkouts/SwiftyGPIO/Sources/SwiftyGPIO.swift:198:60: note: coalesce using '??' to provide a default when the optional value contains 'nil'
            let len = fread(buffer.baseAddress, MAXLEN, 1, fp)
                                                           ^
                                                              ?? <#default value#>
/home/philippe/Desktop/PIRSensorProject/.build/checkouts/SwiftyGPIO/Sources/SwiftyGPIO.swift:198:60: note: force-unwrap using '!' to abort execution if the optional value contains 'nil'
            let len = fread(buffer.baseAddress, MAXLEN, 1, fp)
                                                           ^
                                                             !
error: fatalError`

We're trying to identify the problem, if it is hardware, we'll order a new Pi 5. Or if it is because of Ubuntu, we'll downgrade.. 

Thank you very much, we look forward to being able to use the package.

bottlehall, 2024-08-28:
I had this problem, too. You need to delete the existing lines 175 and 193 in SwiftyGPIO.swift and replace each with::
```swift
guard let fp else { return }
```
Sadly, I suspect this won't get your code working though. See my issue [here](https://github.com/uraimo/SwiftyGPIO/issues/141). I think there have been other changes in recent Ubuntu as well.

FleetPhil, 2024-09-16:
Any thoughts on this? Does downgrading to Ubuntu 22.x make any difference? I'm running on Pi 4...

Alex1478, 2025-02-12:
> Any thoughts on this? Does downgrading to Ubuntu 22.x make any difference? I'm running on Pi 4...

Running into the same issue, have you been able to find a solution?

Alex1478, 2025-02-18:
> Sadly no - it seems like this library is no longer fully functional or supported. The basic GPIO in/out functions are working but not features such as pullup/pulldown or .onChange. I've not been able to find another GPIO library in Swift either....

That's a bummer! Seems like I will need to purchase a Raspberry Pi 4 in this case.
Thank you taking the time to answer